jketterl/openwebrx #92: rtl-sdr fails to change center frequency

jquagga, 2020-03-22:
Good day

So I've found that when I'm changing frequency settings with my rtl-sdr without changing sampling rate or perhaps the direct sampling flag, I get:

```
2020-03-22 11:52:27,517 - owrx.source - DEBUG - activating profile 2m-Low
2020-03-22 11:52:27,518 - owrx.source.connector - DEBUG - sending property change over control socket: center_freq changed to 145150000
rtlsdr_demod_write_reg failed with -9
r82xx_write: i2c wr failed=-9 reg=17 len=1
r82xx_set_freq: failed=-9
csdr shift_addition_cc: reinitialized to -0.245833
```

This results in openwebrx silently loading the gui of the new frequency, but the sdr and the waterfall are actually still displaying the original frequency.  This confused me for example when switching to my local fire department dispatch frequency and hearing the ham bands.  

Now if i change the settings to something which uses direct sampling or a different sample rate, this appears to cause the sdr to "reload" properly and then it works fine.

So here is an example of my config (and I have this issue using rtl_sdr or rtl_sdr_soapy).  If I go from 2M-High to 2M-Low, I get the error above.  If I switch from 2M-High to say, 40M, it re-initalizes properly.  If I then go from 40M to 2M-Low it also re-initializes fine.  So the work around is basically, flip to 40M, and then flip back to 2M-Low.  That works fine.  But going from 2M-High to 2-M-Low or to 70cm doesn't change the center frequency.

Here's the SDR section of my config if that helps.  

```python
sdrs = {
    "rtlsdr": {
        "name": "RTL-SDR USB Stick",
        "type": "rtl_sdr_soapy",
        "ppm": 0,
        "direct_sampling": 0,
        # you can change this if you use an upconverter. formula is:
        # center_freq + lfo_offset = actual frequency on the sdr
        # "lfo_offset": 0,
        "profiles": {
           "2m-High": {
                "name": "2m-High",
                "center_freq": 146500000,
                "rf_gain": 30,
                "samp_rate": 2400000,
                "start_freq": 147300000,
                "start_mod": "nfm",
            },
            "2m-Low": {
                "name": "2m-Low",
                "center_freq": 145150000,
                "rf_gain": 30,
                "samp_rate": 2400000,
                "start_freq": 144390000,
                "start_mod": "nfm",
            },
            "70cm-High": {
                "name": "70cm High",
                "center_freq": 447750000,
                "rf_gain": 30,
                "samp_rate": 2400000,
                "start_freq": 448325000,
                "start_mod": "nfm",
            },
            "70cm-Mid": {
                "name": "70cm Mid",
                "center_freq": 444500000,
                "rf_gain": 30,
                "samp_rate": 2400000,
                "start_freq": 444512500,
                "start_mod": "nfm",
            },
           "70cm-Low": {
                "name": "70cm Low",
                "center_freq": 442500000,
                "rf_gain": 30,
                "samp_rate": 2400000,
                "start_freq": 442512500,
                "start_mod": "nfm",
            },
            "PWCFD": {
                "name": "PWCFD",
                "center_freq": 154000000,
                "rf_gain": 30,
                "samp_rate": 2400000,
                "start_freq": 154570000,
                "start_mod": "nfm",
            },
           "6m": {
                "name": "6m",
                "center_freq": 50500000,
                "rf_gain": 30,
                "samp_rate": 2400000,
                "start_freq": 53490000,
                "start_mod": "nfm",
            },
            "FRS": {
                "name": "FRS",
                "center_freq": 462500000,
                "rf_gain": 30,
                "samp_rate": 2400000,
                "start_freq": 462562500,
                "start_mod": "nfm",
            },
            "11M": {
                "name": "11M-CB",
                "center_freq": 27500000,
                "rf_gain": 30,
                "samp_rate": 2400000,
                "start_freq": 27185000,
                "start_mod": "nfm",
            }
        },
    },
    "rtl_sdr_V3": {
        "name": "RTL-SDR V3 HF+",
        "type": "rtl_sdr_soapy",
        "ppm": 0,
        "direct_sampling": 2,
        "profiles": {
            "20m": {
                "name": "20m",
                "center_freq": 14150000,
                "rf_gain": 10,
                "samp_rate": 768000,
                "start_freq": 14070000,
                "start_mod": "usb",
            },
            "30m": {
                "name": "30m",
                "center_freq": 10125000,
                "rf_gain": 10,
                "samp_rate": 192000,
                "start_freq": 10142000,
                "start_mod": "usb",
            },
            "40m": {
                "name": "40m",
                "center_freq": 7150000,
                "rf_gain": 10,
                "samp_rate": 256000,
                "start_freq": 7070000,
                "start_mod": "lsb",
            },
            "80m": {
                "name": "80m",
                "center_freq": 3650000,
                "rf_gain": 10,
                "samp_rate": 768000,
                "start_freq": 3570000,
                "start_mod": "usb",
            },
            "49m": {
                "name": "49m Broadcast",
                "center_freq": 6000000,
                "rf_gain": 10,
                "samp_rate": 768000,
                "start_freq": 6070000,
                "start_mod": "am",
            },
        },
    },
}
```

This is the latest docker image, c45e74d85bc0.

jketterl, 2020-03-22:
This is an error with the device or the libraries, there's nothing I can do to prevent this. The `rtl_connector` simply calls the `rtlsdr_set_center_freq()` method. Oddly enough, it doesn't even seem to return an error code to the application, if it did you should see a message "WARNING: setting center_freq failed: " with an error code.

I'll admit there's no other error handling in place, but without any error returned from the library, I'm in no position to address this in any way.

jquagga, 2020-03-22:
Yeah, no Warning comes back.  It's easy enough to work around.  Does this only impact rtlsdrs?  I think I'm going to step up to an AirSpy Mini in the near term as I like to use openwebrx to "browse" the airwaves and I think I'd like the wider view I could get with an AirSpy.  And there's also the whole stuck in the house from the virus thing which means I'm using this more.

jketterl, 2020-03-22:
I'd guess it's more hardware related. I just tried to setup hi/low profiles on the 2m range that only differ in the frequency like that:

```
            "2m": {
                "name": "2m komplett",
                "center_freq": 145000000,
                "rf_gain": 30,
                "samp_rate": 2048000,
                "start_freq": 145725000,
                "start_mod": "nfm",
                "lfo_offset": 0,
                "bias_tee": False,
            },
```

that config works fine for me, no errors on the log either.

It might as well be a failure in librtlsdr. This may be an error that has been adressed in recent versions, I wouldn't know, unfortunately.

jquagga, 2020-03-22:
Thank you!  

I'll close this up and bang on the config a bit more later and see if I can figure out anything with it.  It's easy enough to work around for now and I may end up moving to a different SDR in the future anyway.

closed 2020-03-22 by jquagga