jketterl/openwebrx #93: Multiple SDR Dongles on a single Raspy

cybernano96, 2020-03-23:
Hello, i have multiple SDR Keys and i want to use them with different softwares, how can i set the "-d 1" attribute in rtl_tcp, i can't find the string that start the receiver

jketterl, 2020-03-23:
The command is built dynamically, and it doesn't use `rtl_tcp` anymore. You can still configure an identifier on sdr by using the `device` key:

```
sdrs = {
    "rtlsdr": {
        "name": "RTL-SDR USB Stick",
        "type": "rtl_sdr",
        "ppm": 0,
        "device": "1",
        [... other device properties ...]
        "profiles": {
            [...]
```

cybernano96, 2020-03-23:
Thanks, I'll try

cybernano96, 2020-03-23:
it works!

jketterl, 2020-03-25:
awesome! :)

closed 2020-03-25 by jketterl

v01t, 2020-03-30:
@jketterl how to specify remote endpoint address for rtl_tcp like 172.16.25.1:1234?
Thanks.

jketterl, 2020-03-31:
Commenting on closed, unrelated issues is the best way to never get an answer. Please use the mailing list for questions, issues are for bugs.

`rtl_tcp` is not a supported source, you can only use a locally installed stick as a source right now.