jketterl/openwebrx #94: How to select different rtl dongle with multiple in PC

arjanvs, 2020-03-23:
Nice software, but I have two dongles in my system. And one is used for something else. The 2nd or 3rd I like to use for openwebrx. How do I do this?

KR Arjan

jketterl, 2020-03-23:
Well, this is pretty much the same as in https://github.com/jketterl/openwebrx/issues/93 - all you need to do is figure out the device serial numbers, or, if absent, simple use them by index (which would probably be 1 and 2).

takgr, 2020-03-29:
Since there is currently no way to select this from the configuration file, or I didn't find any, I had to modify rtl_sdr.py file in owrx/source adding the -d parameter in rtl_connector:

return super().getCommandMapper().setBase("rtl_connector -d 1")

and it worked

jketterl, 2020-03-29:
there is no need to modify the code :slightly_frowning_face: the `-d` option is mapped to the device key, as described in #93.

```
sdrs = {
    "rtlsdr": {
        "name": "RTL-SDR USB Stick",
        "type": "rtl_sdr",
        "ppm": 0,
        "device": "1",
        [... other device properties ...]
        "profiles": {
            [...]
```

closed 2020-04-19 by jketterl

arjanvs, 2020-08-31:
Sorry for the late response. This clue solved it. But perhaps an idea to put in some documentation?

THX KR Arjan

jketterl, 2020-08-31:
You mean like this? https://github.com/jketterl/openwebrx/wiki/SDR-device-and-profile-configuration#device

arjanvs, 2020-08-31:
Dear Jakob,

It is there, so don't why couldn't find it, when I original wrote the issue. Currently, I don't have openwebrx running on my server, but next time I try to install it again. Sorry for the questions! Keep up the good work!

KR Arjan