jketterl/openwebrx #96: NooElec SDR works but NoElec E4000 does not

Mudary, 2020-04-03:
pi@rpi32:~/openwebrx $ ./openwebrx.py

OpenWebRX - Open Source SDR Web App for Everyone!  | for license see LICENSE file in the package
_________________________________________________________________________________________________

Author contact info:    Jakob Ketterl, DD5JFK <dd5jfk@darc.de>
Documentation:          https://github.com/jketterl/openwebrx/wiki
Support and info:       https://groups.io/g/openwebrx

2020-04-03 14:46:59,616 - owrx.__main__ - INFO - OpenWebRX version v0.19 starting up...
2020-04-03 14:46:59,879 - owrx.sdr - ERROR - The SDR source type "airspyhf" is not available. please check requirements.
2020-04-03 14:47:00,070 - owrx.sdr - ERROR - The SDR source type "sdrplay" is not available. please check requirements.
2020-04-03 14:47:00,072 - owrx.sdr - INFO - SDR sources loaded. Available SDRs: RTL-SDR USB Stick
2020-04-03 14:47:01,500 - owrx.connection - DEBUG - client connection intitialized
2020-04-03 14:47:01,520 - owrx.source - DEBUG - activating profile 70cm
2020-04-03 14:47:01,543 - owrx.source - INFO - Started sdr source: rtl_connector -P 0 -g 30 -s 2400000 -f 438800000 -p 50837 -c 52329
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN:

Using device 0: Generic RTL2832U OEM
Found Elonics E4000 tuner
setting gain failed
2020-04-03 14:47:02,132 - owrx.source - DEBUG - shut down with RC=6
2020-04-03 14:47:02,154 - owrx.source.connector - DEBUG - opening control socket...
2020-04-03 14:47:02,156 - owrx.source - ERROR - Exception during postStart()
Traceback (most recent call last):
  File "/home/pi/openwebrx/owrx/source/__init__.py", line 195, in start
    self.postStart()
  File "/home/pi/openwebrx/owrx/source/connector.py", line 52, in postStart
    self.controlSocket.connect(("localhost", self.controlPort))
ConnectionRefusedError: [Errno 111] Connection refused
2020-04-03 14:47:02,159 - owrx.dsp - DEBUG - received STATE_FAILED, shutting down DspSource

^C2020-04-03 14:47:08,869 - owrx.websocket - DEBUG - websocket loop ended; shutting down
2020-04-03 14:47:08,871 - owrx.websocket - DEBUG - websocket loop ended; sending close frame
2020-04-03 14:47:08,872 - owrx.cpu - DEBUG - cpu usage thread shut down
pi@rpi32:~/openwebrx $

jketterl, 2020-04-03:
This is a configuration issue. The following errors are indicative:

> Found Elonics E4000 tuner
> setting gain failed

The E4000 doesn't support the gain value of 30 that is part of the default config. You'll need to change this to a supported value:

```
Supported gain values (14): -1.0 1.5 4.0 6.5 9.0 11.5 14.0 16.5 19.0 21.5 24.0 29.0 34.0 42.0 
```

unfortunately I cannot change the default configuration in a way so that it will fit all SDRs.

Mudary, 2020-04-03:
Can you please tell me where the default config is, in openwebrx, rtl-sdr or soapysdr? Thank you.

jketterl, 2020-04-03:
The answer is, as usual, it depends, but this is one of the rare cases in which I have some prepared documentation ready :slightly_smiling_face: 

https://github.com/jketterl/openwebrx/wiki/Configuration-guide#where-is-the-configuration-located

Mudary, 2020-04-03:
Thank you, I didn't think of checking the wiki.

closed 2020-04-03 by Mudary