jketterl/openwebrx #97: Center Freq UI Change.

modem2k2, 2020-04-04:
My openwebrx raspi image running very nicely. I'd like to run a tunable VHF/UHF receiver for our local ham group but the Web interface does not allow the receiver center frequency to be changed. I can understand the reason for that where multiple user tune commands could stomp on one another, but in our application that risk is OK.

jketterl, 2020-04-04:
You can change the center frequency. Please have a look at the profiles, the RPi image should come with a profile for both 2m and 70cm frequencies.

modem2k2, 2020-04-04:
That works well for a small number of bands, but the feature I'm interested in and think other people could benefit from is the ability to set the center frequency from the web interface without having to edit the configuration file or having to create dozens of bands.

jketterl, 2020-04-04:
OK so there's a big difference between being able to change the center frequency and giving the user complete freedom to change it, at least if you see it from a legal perspective.

What you're requesting is the so-called "private mode", simply because I cannot recommend making a receiver in a configuration like that public. It's planned, it will need to be explicitly enabled, but it's not top priority and will take some more work.

I'll add the feature flag just to make clear that this is not a bug.

modem2k2, 2020-04-04:
Thanks, certainly is a feature that should be expressly activated in the configuration file, is not intended to openwebrx to be exposed to the general public. But still can be very useful in various scenarios including checking the operation of the receiving station, RFI search, remote monitoring of the proper functioning of communications systems among others.

alexFab, 2020-07-03:
i would appreciate a private mode.

- rf_gain
- center_frequency
- Sample rate 

would be nice.

Old Issue, sounds like someone has an old fork with at least some features like this. https://github.com/ha7ilm/openwebrx/issues/102

Gymnae, 2020-10-28:
That was my fork :) It was a dirty hack and I since migrated to Jakob's fork. I do miss this feature dearly, though. When I want to roam the bands freely I either have to fire up spyserver and temporarily remove the device from openwebrx or I use a kiwisdr based webSDR

m-stefanski, 2020-11-20:
> OK so there's a big difference between being able to change the center frequency and giving the user complete freedom to change it, at least if you see it from a legal perspective.
>
> What you're [...]

Hi, do you have any estimate on this functionality? 
Is there anything a mid python dev can do to help? :)

ale-rinaldi, 2021-07-10:
I think the best way of implementing this feature, to benefit of it also in "public mode" (the current one), is to allow to specify the lower and upper band in a specific profile, and allow the user to move the center in this range.

This would, for example, allow the user to navigate the entire broadcast FM spectrum, not going over that limit.

pilot51, 2022-11-10:
I started using OpenWebRX a couple days ago to replace Gqrx mainly so I could listen on my phone when I'm away from my PC. I am not interested in making it public. It works great, but it is annoying to have to set up profiles for every 2 MHz slice where I might want to listen to something. I'm willing to deal with it for the convenience of listening on my phone.

I would like to be able to select from a list of all bookmarks to jump to it, much like profiles, or quickly tune to any frequency within the limits of the SDR, as most other SDR apps can do.