jketterl/openwebrx #98: sdr.hu/map

UB0ALH, 2020-04-04:
Hello. Why then openwebrx starts in autostart "/home/.../openwebrx/openwebrx-develop/openwebrx.py" it not shows on sdr.hu/map?
But if I start openwebrx from termnal manualy same command it shows well? (

I wish you good health!

jketterl, 2020-04-04:
probably because you have two installations and multiple, different configurations... The command `openwebrx` is only available through the debian packages, but the python code behind that is not contained in `/home`. I don't know what you're doing there, but I'm pretty sure if you start messing with the system it will behave :wink:

UB0ALH, 2020-04-04:
Thanks. I made search for  "config_webrx.py" on disk - there is no other copies of it. There is only one such file in "/home/.../openwebrx/openwebrx-develop/" - I'm confused (((
Are you plan your own "map"?

jketterl, 2020-04-04:
well, there should at least be another one in `/etc/openwebrx/config_webrx.py` - That's the one that is distributed with the package. Also, if your config is in your home directory, you need to make sure that you set the working directory to start `openwebrx.py` from there, since otherwise it won't be able to find that copy.

I won't roll my own map yet, simply because OpenWebRX is a big enough project to keep me more than busy right now. There's a separate issue about that here already.

UB0ALH, 2020-04-04:
I solved problem. I added in autostart "start.sh":

cd /home/.../openwebrx/openwebrx-develop/
./openwebrx.py

UB0ALH, 2020-04-04:
PS: in "/etc/openwebrx/config_webrx.py" the same config

closed 2020-04-04 by UB0ALH

jketterl, 2020-04-04:
Well, the recommended way of having OpenWebRX start automatically is to use systemd. The debian package automatically handles that for you, and there are instructions available on how to set it up for manual installations: https://github.com/jketterl/openwebrx/wiki/Manual-Package-installation-(including-digital-voice)#installing-openwebrx-as-a-systemd-service

Other mechanisms may of course work too, but I can't really support all of them.

Also, maintaining a `config_webrx.py` in the checkout should only be used for development. The official location is `/etc/openwebrx/config.py`.