jketterl/openwebrx #99: Error 503 Backend fetch failed

2E0LXY, 2020-04-06:
Error 503 Backend fetch failed
Backend fetch failed
Guru Meditation:
XID: 6
Varnish cache server

Any ideas please?

jketterl, 2020-04-06:
This error does not allow a real diagnosis since it only shows that OpenWebRX is not running.

Please check the locks with `sudo journalctl -u openwebrx`.

2E0LXY, 2020-04-06:
[journalctl .txt](https://github.com/jketterl/openwebrx/files/4440061/journalctl.txt)

jketterl, 2020-04-06:
> Apr 06 17:49:19 openwebrx openwebrx[781]:   File "/etc/openwebrx/config_webrx.py", line 1, in <module>
> Apr 06 17:49:19 openwebrx openwebrx[781]:     c# -*- coding: utf-8 -*-

There's a spurious `c` right there at the beginning of your config file, which leads to the config being unreadable.

2E0LXY, 2020-04-06:
:-)
Thanks for that!
http://2e0lxy.dynu.net:70

Just so many questions and so much to read! HA

jketterl, 2020-04-06:
no worries :) have fun playing :)

closed 2020-04-06 by jketterl