jketterl/openwebrx #100: decoding digital

DrHyperKALICH, 2020-04-07:
Hello!

mbelib is installed 

Install the project...
-- Install configuration: ""
-- Up-to-date: /usr/local/lib/pkgconfig/libmbe.pc
-- Up-to-date: /usr/local/lib/libmbe.a
-- Up-to-date: /usr/local/include/mbelib.h
-- Up-to-date: /usr/local/lib/libmbe.so.1.3
-- Up-to-date: /usr/local/lib/libmbe.so.1
-- Up-to-date: /usr/local/lib/libmbe.so
-- Up-to-date: /usr/local/include/mbelib.h

digiham is installed

Install the project...
-- Install configuration: ""
-- Up-to-date: /usr/local/bin/rrc_filter
-- Up-to-date: /usr/local/bin/dc_block
-- Up-to-date: /usr/local/bin/gfsk_demodulator
-- Up-to-date: /usr/local/bin/dmr_decoder
-- Up-to-date: /usr/local/bin/ysf_decoder
-- Up-to-date: /usr/local/bin/mbe_synthesizer
-- Up-to-date: /usr/local/bin/digitalvoice_filter
-- Up-to-date: /usr/local/bin/fsk_demodulator
-- Up-to-date: /usr/local/bin/pocsag_decoder

BUT i getting

mbe_synthesizer: error while loading shared libraries: libmbe.so.1: cannot open shared object file: No such file or directory

May be I need too put this libs to specific folder?

jketterl, 2020-04-07:
This looks more like a distribution-specific problem with the shared library lookup. First of all, run `ldconfig`, maybe it's just a matter of updating the cache.

Second, see if the library folders that are used by mbelib (`/usr/local/lib`) are listed in your shared library lookup config. This is a distribution specific task, so I cannot give precise instructions, but usually the config resides in `/etc/ld.so.conf` or `/etc/ld.so.conf.d`.

DrHyperKALICH, 2020-04-07:
Solved by copying mbelib libs to /usr/lib/

closed 2020-04-07 by DrHyperKALICH