jketterl/openwebrx #101: Finding config file , config_webrx.py

La8gka, 2020-04-09:
Hello
Thanks for the excellent work with the Openwebrx
I have downloaded the image for rsp3 and a RTLSDR dongle V3 and after some
strugle i managed to get the openwebrx page up running with all rtl-sdr features to YES

But i cant find the config_webrx.py file , im a newbee on linux so i need som help
im trying the nano ../openwebrx/config_webrx.py command but im for sure doing something wrong ?
[image]

jketterl, 2020-04-09:
Please see https://github.com/jketterl/openwebrx/wiki/Configuration-guide#where-is-the-configuration-located

La8gka, 2020-04-09:
Hi thanks 
been there and also tried the /etc/openwebrx . But i cant find the openwebrx folder anywhere
Im getting one fail at startup "Failed to start Varnish HTTP Accelerator log daemon" , the rest is "OK"
Im logged directly locally in to the Raspberry pi at:
Openwebrx login:pi
password:raspberry
pi@openwebrx:

Is there another way of logging in ?

jketterl, 2020-04-09:
Well, are you looking for the config file or the openwebrx folder? I feel like there's a lot of things happening at once.

You can log into your receiver using SSH after you have enabled it in the `raspi-config` utility. That's just as when using Raspbian, I suppose it's a good idea to familiarize yourself with it since the images are built on the same base.

The configuration file is `/etc/openwebrx/config_webrx.py` on the Raspberry Pi images. If that file is not there, something bad has happened during the image transfer onto the SD card, or at a later point. The correct command to edit the configuration would be `sudo nano /etc/openwebrx.config_webrx.py`.

Looking at the screenshot, I have a feeling you're having more problems. The fact that varnish failed to start is not a good sign either.

hamradio95, 2020-04-10:
Hi,

I have got the same pb, when i'm looking you screeshot, you can see that the number of maximum connection is full (red indication 4/4)

The log connection tell me, The WS connection try, and fail. Many user is show, but it was the number of re trying. WS is refused and create multi connection/tentative user...

Excuse me for my bad english written, and thank you for your work !!!!!!

jketterl, 2020-04-10:
All that signifies hat *some* error is happening, but there's no way telling if it's the same. You'll need to check the logs for errors. And then please open your own issue, or post to the mailing list for help.

hamradio95, 2020-04-10:
thank you jketterl for you answer.
and sorry, i'm just start linux, it's hard for me. I'm trying to find the log, and be back.
Do you prefer that a new post, or can i use this one ?

best number to you and you family !

La8gka, 2020-04-10:
sudo nano /etc/openwebrx/config_webrx.py is working fine, thanks jketterl
but still some strange tings,decided to start over again with fresh SD install

-Loaded Raspberry image on SD card
- startet Raspberry 3 with RTL_SDr dongle V3 , openwebrx web says Error Message:
No SDR Devices available
Feature check says NO on rtl_sdr and rtl_connector
installed the corresponding header files : sudo apt-get install librtlsdr-dev
Stil No SDR Devices available

Installed SoapyRTLSDR:

git clone https://github.com/pothosware/SoapyRTLSDR.git
cd SoapyRTLSDR
mkdir build
cd build
cmake ..
make
sudo make install

Installed owrx_connector from source:

git clone https://github.com/jketterl/owrx_connector.git
cd owrx_connector
mkdir build
cd build
cmake ..
make
sudo make install

Features is now YES
But still not working,what im i doing wrong ?

Logfile
-- Logs begin at Fri 2020-04-10 22:43:11 UTC, end at Fri 2020-04-10 22:50:35 UTC. --
Apr 10 22:49:10 openwebrx systemd[1]: Started OpenWebRX WebSDR receiver.
Apr 10 22:49:22 openwebrx openwebrx[658]: 2020-04-10 22:49:22,943 - owrx.sdr - INFO - SDR sources loaded. Available SDRs: RTL-SDR USB Stick, Airspy HF+, SDRPlay RSP2
Apr 10 22:49:27 openwebrx openwebrx[658]: 2020-04-10 22:49:27,201 - owrx.connection - DEBUG - client connection intitialized
Apr 10 22:49:27 openwebrx openwebrx[658]: 2020-04-10 22:49:27,286 - owrx.source - DEBUG - activating profile 70cm
Apr 10 22:49:27 openwebrx openwebrx[658]: 2020-04-10 22:49:27,328 - owrx.source - DEBUG - activating profile 20m
Apr 10 22:49:27 openwebrx openwebrx[658]: 2020-04-10 22:49:27,348 - owrx.source - DEBUG - activating profile 20m
Apr 10 22:49:27 openwebrx openwebrx[658]: 2020-04-10 22:49:27,401 - owrx.source - INFO - Started sdr source: rtl_connector -s 2400000 -P 0 -g 30 -f 438800000 -p 55613 -c 47429
Apr 10 22:49:27 openwebrx openwebrx[658]: Found 1 device(s):
Apr 10 22:49:27 openwebrx openwebrx[658]:   0:  Realtek, RTL2838UHIDIR, SN: 00000001
Apr 10 22:49:27 openwebrx openwebrx[658]: Using device 0: Generic RTL2832U OEM
Apr 10 22:49:27 openwebrx openwebrx[658]: Detached kernel driver
Apr 10 22:49:28 openwebrx openwebrx[658]: Found Rafael Micro R820T tuner
Apr 10 22:49:28 openwebrx openwebrx[658]: [R82XX] PLL not locked!
Apr 10 22:49:28 openwebrx openwebrx[658]: IQ worker thread started
Apr 10 22:49:28 openwebrx openwebrx[658]: socket setup complete, waiting for connections
Apr 10 22:49:28 openwebrx openwebrx[658]: setting up control socket...
Apr 10 22:49:28 openwebrx openwebrx[658]: control socket started on 47429
Apr 10 22:49:28 openwebrx openwebrx[658]: Allocating 2 zero-copy buffers
Apr 10 22:49:28 openwebrx openwebrx[658]: 2020-04-10 22:49:28,554 - owrx.source.connector - DEBUG - opening control socket...
Apr 10 22:49:28 openwebrx openwebrx[658]: client connection establised
Apr 10 22:49:28 openwebrx openwebrx[658]: control connection established
Apr 10 22:49:28 openwebrx openwebrx[658]: 2020-04-10 22:49:28,573 - owrx.dsp - DEBUG - received STATE_RUNNING, attempting DspSource restart
Apr 10 22:49:28 openwebrx openwebrx[658]: 2020-04-10 22:49:28,588 - csdr.csdr - DEBUG - Command = nc -v 127.0.0.1 55613 | csdr shift_addition_cc --fifo /tmp/openwebrx/openwebrx_pipe_3041693648_shift_pipe | csdr fir_decimate_cc 217 0.0006
Apr 10 22:49:28 openwebrx openwebrx[658]: 2020-04-10 22:49:28,655 - owrx.dsp - DEBUG - adding new output of type audio
Apr 10 22:49:28 openwebrx openwebrx[658]: closing client socket
Apr 10 22:49:29 openwebrx openwebrx[658]: Illegal instruction
Apr 10 22:49:29 openwebrx openwebrx[658]: Illegal instruction
Apr 10 22:49:29 openwebrx openwebrx[658]: Illegal instruction
Apr 10 22:49:29 openwebrx openwebrx[658]: Illegal instruction
Apr 10 22:49:29 openwebrx openwebrx[658]: Illegal instruction
Apr 10 22:49:29 openwebrx openwebrx[658]: Illegal instruction
Apr 10 22:49:29 openwebrx openwebrx[658]: Illegal instruction
Apr 10 22:49:29 openwebrx openwebrx[658]: Illegal instruction
Apr 10 22:49:29 openwebrx openwebrx[658]: Illegal instruction
Apr 10 22:49:29 openwebrx openwebrx[658]: Connection to 127.0.0.1 55613 port [tcp/*] succeeded!
Apr 10 22:49:29 openwebrx openwebrx[658]: client connection establised
Apr 10 22:49:29 openwebrx openwebrx[658]: Illegal instruction
Apr 10 22:49:29 openwebrx openwebrx[658]: 2020-04-10 22:49:29,192 - csdr.csdr - DEBUG - dsp thread ended with rc=132
Apr 10 22:49:29 openwebrx openwebrx[658]: closing client socket

jketterl, 2020-04-11:
Wow, this looks like something really really bad is happening, I won't even go down the list. First thing I want to work is this:

> Feature check says NO on rtl_sdr and rtl_connector

I already stated that the images come preinstalled with all the requirements for rtl_sdr, so this should absolutely not be the case. All your attempts to manually install things after that should not be necessary, and to make things worse, will probably cause problems further down the road when the next update is due.

Just to clear out some assumptions:

- What image are you using and where did you get it from?
- What is your process for flashing the image to the card?
- Did you in any way modify the card or image during the whole process?

You say you're using a Raspberry Pi 3, is it in any way possible this is not a genuine model? I do see that there's "Illegal instruction" errors in your log, probably coming from csdr. The images however should contain binaries that have been specifially compiled for the CPUs of the various RPi generations, and I'm very surprised to see it in there. Would you mind sharing the contents of `/proc/cpuinfo`?

La8gka, 2020-04-11:
Hi again 
•	What image are you using and where did you get it from?
https://www.openwebrx.de/download/rpi.php  2020-03-21-OpenWebRX-full.zip

•	What is your process for flashing the image to the card?
Flashing the 16GB Class10 card with Raspberry Pi Imager, and then putting it directly in the Pi3

•	Did you in any way modify the card or image during the whole process?
No

Feature check says NO on rtl_sdr and rtl_connector after this process (the SDRRTL is plugged in)

Actually i see in the /proc/cpuinfo that it is a Pi2 ,i was sure it was a Pi3
That could be the problem ?   If so i will order a Pi4

processor       : 0
model name      : ARMv6-compatible processor rev 7 (v6l)
BogoMIPS        : 697.95
Features        : half thumb fastmult vfp edsp java tls
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xb76
CPU revision    : 7

Hardware        : BCM2835
Revision        : 000e
Serial          : 00000000eaad0133
Model           : Raspberry Pi Model B Rev 2

jketterl, 2020-04-11:
Thanks for your details. Nothing wrong for the most part, as I said, just wanted to clar out assumptions.

As for your Raspberry... that's not even a Pi 2, it's a Pi 1, single-core. Not sure if that has enough grunt to be enjoyable, but I do have one of them available. I'll try running the image.

La8gka, 2020-04-11:
Hi
Borrowed a Raspberry 3 from a friend, flashed the image and worked right away. Thank you for your help and patience Jakob
I will by a Raspberry 4 for this project
73 Rune LA8GKA

jketterl, 2020-04-11:
Well now, I spent some time and tried the image out with the various Raspberry Pi models I have at my hands right now. It doesn't work in the Raspberry Pi 1 and the Raspberry Pi Zero since they are still using armv6 generation CPUs. Unfortunately, the debian packaging system does not separate those, they are both considered "armhf".

I'm not sure if there's a good way to solve this, but during the process I also saw that a Raspberry Pi 2 is not capable of running a single client smoothly at 2.4 MS/s, so I'm pretty much expecting that a Raspberry Pi 1 will be mostly useless. So if you have a chance to upgrade the hardware, I'd strongly recommend you go that way.

I will probably need to put up some documentation about this.

closed 2020-05-03 by jketterl