jketterl/openwebrx #102: Error on FT8 decoding that persists over restart

dh5ym, 2020-04-09:
After some 80m FT8 decoding i found that the FT8 decoder was not working anymore. The issue is persistant on restart. 
Each time the FT8 decoder is invoked it will cause a SIGABRT, for example:

fftw: planner.c:891: assertion failed: flags.u == u

Program received signal SIGABRT: Process abort signal.

Backtrace for this error:
fftw: planner.c:891: assertion failed: flags.u == u

Program received signal SIGABRT: Process abort signal.

Backtrace for this error:
#0  0x7fe335ccf32a
#1  0x7fe335cce503
#2  0x7fe334d93f1f
#3  0x7fe334d93e97
#4  0x7fe334d95800
#5  0x7fe3360b5915
#6  0x7fe3360b77c4
#7  0x7fe33617f7a3
#8  0x7fe33617f7e1
#9  0x55a89ecb8189
#10  0x55a89ecb6dce
#11  0x7fe334d76b96
#12  0x55a89ecb6ec9
#13  0xffffffffffffffff
#0  0x7fbb1a8b432a
#1  0x7fbb1a8b3503
#2  0x7fbb19978f1f
#3  0x7fbb19978e97
#4  0x7fbb1997a800
#5  0x7fbb1ac9a915
#6  0x7fbb1ac9c7c4
#7  0x7fbb1ad647a3
#8  0x7fbb1ad647e1
#9  0x55ede40fa189
#10  0x55ede40f8dce
#11  0x7fbb1995bb96
#12  0x55ede40f8ec9
#13  0xffffffffffffffff
2020-04-09 22:14:15,325 - owrx.wsjt - WARNING - decoder return code: -6

jketterl, 2020-04-09:
I'm afraid that's a failure in WSJT-X. Don't think there's much I can do about this.

dh5ym, 2020-04-09:
I will try to find the way to get it working again here and will write it down here for documentation purpose.

dh5ym, 2020-04-09:
i deleted /tmp/jt9_wisdom.dat which seams to resolve the issue.
Thanks for your help.

closed 2020-04-09 by dh5ym