jketterl/openwebrx #103: Openwebrx and rtl_tcp

Excalibur201010, 2020-04-10:
Good day.
I have multiple rtl-sdr dongles located i many places and connected to the websdr, i want to migrate to Openwebrx, so can it connected to the rtl_tcp?
Sorry for my English.
Have a nice day 73!!!

jketterl, 2020-04-19:
Hello

`rtl_tcp` is currently not supported as a source. It shouldn't be that difficult, but I will need to write a new connector for it.

What's available already is [SoapyRemote](https://github.com/pothosware/SoapyRemote/wiki), so you could have a look into that. It does support rtlsdr sticks, so should theoretically work.

73s
Jakob DD5JFK

Excalibur201010, 2020-04-19:
Okay thanks for answer, i trying soapyRemote =)
73!!! US1GHQ

jketterl, 2020-08-16:
I've implemented the new "rtl_tcp_connector", along with the new source type "rtl_tcp" in owrx_connector and OpenWebRX today. I'm currently still testing myself, but it seems to be mostly working. It's available in the experimental repositories, should be in the next release.

Excalibur201010, 2020-08-16:
Okay im trying tomorow testing =)
73! Best regards
US1GHQ - Alexsandr

Excalibur201010, 2020-08-17:
Good day, Successfully migrating my SDRs from Soapy Remote to rtl_tcp, working perfectly.
Starting testing =)

Excalibur201010, 2020-08-20:
After several days, all working fine.
Thanks and 73!

closed 2020-08-20 by Excalibur201010