jketterl/openwebrx #105: SDRPLay ZIF/LIF parameters in profiles section of config_webrx.py

jshuggins, 2020-04-17:
Hello,

Apologies if I missed where to do this, but is there a way to expose and set the SDRPlay's LIF/ZIF settings so config_webrx.py?

Thanks.

jketterl, 2020-04-19:
Hello,

I suppose you are referring to these settings here:

```
     * IF Mode - IF frequency in kHz
       [key=if_mode, default=Zero-IF, type=string, options=(Zero-IF, 450kHz, 1620kHz, 2048kHz)]
```

if so it shouldn't be difficult to include a corresponding mapping in the next version.

jshuggins, 2020-04-20:
That's great. Thanks!

jketterl, 2020-05-03:
Completely forgot to post about this, but the latest dev builds have this now, the corresponding config key is `if_mode`. I haven't done any testing myself since I have yet to understand what it's even useful for :grin:

jshuggins, 2020-05-04:
Thanks. Will give this a test sometime this week.

jketterl, 2020-05-24:
Any updates on this? Did it work?

jshuggins, 2020-05-27:
I'm still busy using your system for a bunch of testing Jakob, but will get on this soonest.

closed 2020-08-26 by jketterl