jketterl/openwebrx #107: Stale .wav files eventually fill /tmp/openwebrx directory

jshuggins, 2020-04-21:
I have my R-Pi 3B+ setup in config_webrx.py with many bands including the defaults. Some of the bands do not include frequencies in need of decoding via WSPR, FT8, etc.

Scenario:
When I am not using the SDR via browser, the scheduler selects a band to decode. All is well.  When I visit the SDR and select a band that has frequencies to decode, all is well. Then when I visit another band that does not have any frequencies to decode, often a wav file that was in process is left behind to become an orphan... never to be deleted again.

Eventually /tmp/openwebrx fills and causes problems and big /var/syslog files full of /tmp system write errors.

Example stale .wav files below...

pi@openwebrx:/tmp/openwebrx $ ls -l
total 4280
-rw-r--r-- 1 openwebrx openwebrx    8277 Apr 21 01:18 ALL_WSPR.TXT
-rw-r--r-- 1 openwebrx openwebrx    2418 Apr 21 01:18 decoded.txt
-rw-r--r-- 1 openwebrx openwebrx     452 Apr 21 01:18 hashtable.txt
-rw-r--r-- 1 openwebrx openwebrx   12289 Apr 21 01:18 jt9_wisdom.dat
prw-r--r-- 1 openwebrx openwebrx       0 Apr 21 00:58 openwebrx_pipe_1957366928_bpf_pipe
prw-r--r-- 1 openwebrx openwebrx       0 Apr 21 01:10 openwebrx_pipe_1957366928_shift_pipe
prw-r--r-- 1 openwebrx openwebrx       0 Apr 21 01:20 openwebrx_pipe_1957366928_smeter_pipe
prw-r--r-- 1 openwebrx openwebrx       0 Apr 21 00:58 openwebrx_pipe_1957366928_squelch_pipe
-rw-r--r-- 1 openwebrx openwebrx 2875436 Apr 21 01:17 openwebrx-wsjtchopper-1956955152-200421_0116.wav
-rw-r--r-- 1 openwebrx openwebrx  364588 Apr 21 01:17 openwebrx-wsjtchopper-1956997712-200421_011745.wav
-rw-r--r-- 1 openwebrx openwebrx  178220 Apr 21 01:18 openwebrx-wsjtchopper-1956997872-200421_011800.wav
-rw-r--r-- 1 openwebrx openwebrx  358444 Apr 21 01:11 openwebrx-wsjtchopper-1957538576-200421_011130.wav
-rw-r--r-- 1 openwebrx openwebrx  356396 Apr 21 00:45 openwebrx-wsjtchopper-1967230192-200421_004530.wav
-rw-r--r-- 1 openwebrx openwebrx  178220 Apr 21 00:45 openwebrx-wsjtchopper-1967281424-200421_004537.wav
-rw-r--r-- 1 openwebrx openwebrx     690 Apr 21 01:18 timer.out
-rw-r--r-- 1 openwebrx openwebrx     370 Apr 21 01:18 wspr_spots.txt
-rw-r--r-- 1 openwebrx openwebrx     409 Apr 21 01:18 wspr_timer.out
-rw-r--r-- 1 openwebrx openwebrx    2117 Apr 21 01:18 wspr_wisdom.da

"sudo systemctl restart openwebrx.service" doesn't remedy. Not saying it should either... just an observation.

A reboot reestablishes normalcy.

Perhaps this is all perfectly normal. Just don't know.

jketterl, 2020-04-21:
This has been pointed out on the mailing list before, unfortunately without leads. This looks a little more promising, I will try to reproduce it.

It certainly isn't intended to leave those files behind. I have already tried to cover as many edge cases as I could think of, but obviously I missed some.

BTW: It's fixed by a reboot because `/tmp/openwebrx` is an in-memory filesystem. I have had mixed results clearing it out manually while openwebrx is running, but there's no problem clearing the contents while it is stopped.

jshuggins, 2020-04-21:
Thank you.

For the moment, I have this crontab entry to keep an eye in things...

*/10 * * * *	sudo find /tmp/openwebrx -type f -mmin +10 -exec rm -f {} + >/dev/null 2>&1

jketterl, 2020-05-21:
While moving some of this stuff for JS8, I found a hole in the error handling. Basically, if there was exceptions happening during certain parts of the decoding, it would break the flow and leave the file. This has been adressed by now, I'm not sure it's 100% watertight, but it made an improvement.

During the process, however, I also found that the background decoding services were not correctly setting up the temporary directory, and now I'm finding lots of fifos left behind...

Also, there's still the wsjt-x binaries that are polluting the temporary folder with files that have garbage names. The contents seem to be fftw optimization tables, so it may help the performance to get those straights, but I haven't figured out why this keeps happening...

jshuggins, 2020-05-27:
Okay I've now brought my experimental R-Pi running OpenWebRX up to the experimental version and am changing bands via the scheduler every ten minutes.  On the older version the band switch seemed to leave some stale files.  Am watching the new version to see what happens.

jketterl, 2020-07-25:
I haven't been able to reproduce this issue on the latest versions for some time now. I still get occasional pipes left behind, as well as the cryptically named files by jt9 / js8... but the wavefile handling seems to be alright now, so closing this.

closed 2020-07-25 by jketterl