jketterl/openwebrx #108: Big daemon.log due in part to missing avatar image?

jshuggins, 2020-04-21:
Spot reporting seems to work very very well...

  https://pskreporter.info/pskmap.html?preset&callsign=kx4o&txrx=rx&timerange=3600&hideunrec=1&blankifnone=1&hidepink=1

I do wonder about this many-line report in daemon.log that seems to center on a missing png file at htdocs/gfx/openwebrx-avatar.png.

Apr 21 14:21:49 openwebrx openwebrx[710]: 2020-04-21 14:21:49,834 - owrx.pskreporter - DEBUG - uploading 146 spots
Apr 21 14:21:49 openwebrx openwebrx[710]: 2020-04-21 14:21:49,892 - owrx.pskreporter - DEBUG - scheduling next pskreporter upload in 319.614575 seconds
Apr 21 14:25:04 openwebrx openwebrx[710]: 2020-04-21 14:25:04,955 - owrx.sdrhu - INFO - Update succeeded!
Apr 21 14:27:09 openwebrx openwebrx[710]: 2020-04-21 14:27:09,508 - owrx.pskreporter - DEBUG - uploading 160 spots
Apr 21 14:27:09 openwebrx openwebrx[710]: 2020-04-21 14:27:09,557 - owrx.pskreporter - DEBUG - scheduling next pskreporter upload in 314.162813 seconds
Apr 21 14:32:23 openwebrx openwebrx[710]: ----------------------------------------
Apr 21 14:32:23 openwebrx openwebrx[710]: Exception happened during processing of request from ('167.99.214.222', 45376)
Apr 21 14:32:23 openwebrx openwebrx[710]: Traceback (most recent call last):
Apr 21 14:32:23 openwebrx openwebrx[710]:   File "/usr/lib/python3.7/socketserver.py", line 650, in process_request_thread
Apr 21 14:32:23 openwebrx openwebrx[710]:     self.finish_request(request, client_address)
Apr 21 14:32:23 openwebrx openwebrx[710]:   File "/usr/lib/python3.7/socketserver.py", line 360, in finish_request
Apr 21 14:32:23 openwebrx openwebrx[710]:     self.RequestHandlerClass(request, client_address, self)
Apr 21 14:32:23 openwebrx openwebrx[710]:   File "/usr/lib/python3/dist-packages/owrx/http.py", line 26, in __init__
Apr 21 14:32:23 openwebrx openwebrx[710]:     super().__init__(request, client_address, server)
Apr 21 14:32:23 openwebrx openwebrx[710]:   File "/usr/lib/python3.7/socketserver.py", line 720, in __init__
Apr 21 14:32:23 openwebrx openwebrx[710]:     self.handle()
Apr 21 14:32:23 openwebrx openwebrx[710]:   File "/usr/lib/python3.7/http/server.py", line 426, in handle
Apr 21 14:32:23 openwebrx openwebrx[710]:     self.handle_one_request()
Apr 21 14:32:23 openwebrx openwebrx[710]:   File "/usr/lib/python3.7/http/server.py", line 414, in handle_one_request
Apr 21 14:32:23 openwebrx openwebrx[710]:     method()
Apr 21 14:32:23 openwebrx openwebrx[710]:   File "/usr/lib/python3/dist-packages/owrx/http.py", line 32, in do_GET
Apr 21 14:32:23 openwebrx openwebrx[710]:     self.router.route(self)
Apr 21 14:32:23 openwebrx openwebrx[710]:   File "/usr/lib/python3/dist-packages/owrx/http.py", line 76, in route
Apr 21 14:32:23 openwebrx openwebrx[710]:     controller(handler, request).handle_request()
Apr 21 14:32:23 openwebrx openwebrx[710]:   File "/usr/lib/python3/dist-packages/owrx/controllers.py", line 59, in handle_request
Apr 21 14:32:23 openwebrx openwebrx[710]:     "avatar_ctime": os.path.getctime("htdocs/gfx/openwebrx-avatar.png"),
Apr 21 14:32:23 openwebrx openwebrx[710]:   File "/usr/lib/python3.7/genericpath.py", line 65, in getctime
Apr 21 14:32:23 openwebrx openwebrx[710]:     return os.stat(filename).st_ctime
Apr 21 14:32:23 openwebrx openwebrx[710]: FileNotFoundError: [Errno 2] No such file or directory: 'htdocs/gfx/openwebrx-avatar.png'
Apr 21 14:32:23 openwebrx openwebrx[710]: ----------------------------------------
Apr 21 14:32:23 openwebrx openwebrx[710]: 2020-04-21 14:32:23,725 - owrx.pskreporter - DEBUG - uploading 192 spots
Apr 21 14:32:23 openwebrx openwebrx[710]: 2020-04-21 14:32:23,789 - owrx.pskreporter - DEBUG - scheduling next pskreporter upload in 328.495367 seconds

jketterl, 2020-04-21:
Thank you, very interesting report. This seems to be coming from the sdr.hu integration... Background is that sdr.hu downloads the `/status` url of all registered receivers in a scheduled interval, and the response contains the timestamp of the avatar image. I suppose that's to tell sdr.hu that the image has been updated without downloading the actual image.

What's broken is that the file location is still hardcoded, which is broken when using packaged installs.

I will look into this, shouldn't be hard to fix.

jketterl, 2020-04-22:
I tested this out yesterday with an experimental build on a RPi. Are you using the experimental repository and could you verify?

jshuggins, 2020-04-22:
I am not using the experimental repository, rather using the R-Pi ISO image. I would be delighted to convert to the repository version. Will need to figure that one out.  Diffs to particular files work as well if all else fails.

jketterl, 2020-04-22:
If you're using one of the images that was released with 0.18.0 (2020-02-20 or 2020-03-21), it is already built on the repository, but only the "stable" one. That way you will be able to update once a new release comes out.

The experimental repository contains automated development builds, but it's not recommended for production use. You can find more information [here](https://www.openwebrx.de/download/debian.php).

But please don't misunderstand: I don't want to put pressure on you to switch just to test this. We can verify this with the next release, no hurry here :slightly_smiling_face:

jshuggins, 2020-04-23:
I'm all in for using the development branch. To get inline with this brach can I just use the details you provided here...

  https://www.openwebrx.de/download/debian.php

...to my R-Pi image I flashed from the download or should I just pave it with a fresh Raspbian install and add OpenWebRX via github proper?

jketterl, 2020-04-23:
You can use the instructions, the first part should already be in place, so you'll just need to follow the instructions to add the experimental repo. After that, every time you run your apt upgrades, the system should download the latest development builds.

jshuggins, 2020-04-23:
Perfect. I think install is complete. I'll test things out in a hour or so after my "COVID-19 work at home" tele-cons are complete. Thanks!

jshuggins, 2020-04-23:
I put it on 20m where there are spots a plenty to be found using the "UNDER CONSTRUCTION" code pull. Things look more peaceful now...

Apr 23 21:44:53 openwebrx openwebrx[9361]: Connection to 127.0.0.1 43011 port [tcp/*] succeeded!
Apr 23 21:44:53 openwebrx openwebrx[9361]: nmux: pthread_create() done, clients now: 3
Apr 23 21:44:53 openwebrx openwebrx[9361]: client 0xe43c68: started!
Apr 23 21:44:53 openwebrx openwebrx[9361]: csdr tee: file opened: /tmp/openwebrx_pipe_1976148272_iqtee2_pipe
Apr 23 21:44:53 openwebrx openwebrx[9361]: csdr bandpass_fir_fft_cc: (fft_size = 512) = (taps_length = 149) + (input_size = 364) - 1
Apr 23 21:44:53 openwebrx openwebrx[9361]: (overlap_length = 148) = taps_length - 1
Apr 23 21:44:53 openwebrx openwebrx[9361]: csdr shift_addition_cc: reinitialized to -0.225
Apr 23 21:44:53 openwebrx openwebrx[9361]: csdr bandpass_fir_fft_cc: filter initialized, low_cut = 0.1125, high_cut = 0.1375
Apr 23 21:44:56 openwebrx systemd[1]: Started Session c7 of user pi.
Apr 23 21:45:00 openwebrx openwebrx[9361]: 2020-04-23 21:45:00,748 - owrx.pskreporter - DEBUG - scheduling next pskreporter upload in 328.078560 seconds
Apr 23 21:47:57 openwebrx openwebrx[9361]: 2020-04-23 21:47:57,664 - owrx.sdrhu - INFO - Update succeeded!
Apr 23 21:50:28 openwebrx openwebrx[9361]: 2020-04-23 21:50:28,830 - owrx.pskreporter - DEBUG - uploading 326 spots
Apr 23 21:50:28 openwebrx openwebrx[9361]: 2020-04-23 21:50:28,917 - owrx.pskreporter - DEBUG - scheduling next pskreporter upload in 310.824394 seconds

I'll keep an eye on things.

jshuggins, 2020-04-24:
I think the latest pull from the development branch pulled earlier today has successfully addressed this problem. Thank you.

closed 2020-04-24 by jshuggins