jketterl/openwebrx #109: What wrong with me?

d9394, 2020-04-22:
Hi, jketterl:
   After git pull , my openwebrx can not run up , what wrong with me:

2020-04-22 22:15:41,407 - owrx.__main__ - INFO - OpenWebRX version v0.19 starting up...
Traceback (most recent call last):
  File "openwebrx.py", line 6, in <module>
    main()
  File "/opt/openwebrx/owrx/__main__.py", line 60, in main
    SdrService.loadProps()
  File "/opt/openwebrx/owrx/sdr.py", line 45, in loadProps
    name: loadIntoPropertyManager(value) for (name, value) in pm["sdrs"].items() if sdrTypeAvailable(value)
  File "/opt/openwebrx/owrx/sdr.py", line 45, in <dictcomp>
    name: loadIntoPropertyManager(value) for (name, value) in pm["sdrs"].items() if sdrTypeAvailable(value)
  File "/opt/openwebrx/owrx/sdr.py", line 29, in sdrTypeAvailable
    if not featureDetector.is_available(value["type"]):
  File "/opt/openwebrx/owrx/feature.py", line 68, in is_available
    return self.has_requirements(self.get_requirements(feature))
  File "/opt/openwebrx/owrx/feature.py", line 79, in has_requirements
    passed = passed and self.has_requirement(requirement)
  File "/opt/openwebrx/owrx/feature.py", line 91, in has_requirement
    return method()
  File "/opt/openwebrx/owrx/feature.py", line 293, in has_soapy_airspyhf
    return self._has_soapy_driver("airspyhf")
  File "/opt/openwebrx/owrx/feature.py", line 255, in _has_soapy_driver
    drivers = [s.strip() for s in matches[1].split(", ")]
TypeError: '_sre.SRE_Match' object is not subscriptable

[1]+  Exit 1                  python3 -u openwebrx.py

jketterl, 2020-04-22:
what's your python3 version? should be at least python 3.5

d9394, 2020-04-22:
root@DietPi:/opt/openwebrx# SoapySDRUtil --info
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.8.0-gf722f9ce
API Version: v0.8.0
ABI Version: v0.8
Install root: /usr/local
Search path:  /usr/local/lib/SoapySDR/modules0.8
Module found: /usr/local/lib/SoapySDR/modules0.8/libRedPitaya.so      (0.1.1-3d576f8)
Module found: /usr/local/lib/SoapySDR/modules0.8/libsdrPlaySupport.so (0.3.0-14ec39e)
Available factories... redpitaya, sdrplay
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]
root@DietPi:/opt/openwebrx# python3 --version
Python 3.5.3

d9394, 2020-04-22:
After I modify the owrx/feature.py of line 255:

drivers = [s.strip() for s in matches.group(1).split(", ")]

then it works.

jketterl, 2020-04-22:
I remember that I've had to to such modifications before, so reopening just for me to include this :)

d9394, 2020-04-22:
And , if the redpitaya support is not complete? how about the configure of redpitaya in config_webrx.py?

jketterl, 2020-04-22:
Well... it's not complete since in the only test we did, it didn't work. The soapy implementation was unable to deliver the data chunks in sizes that the connector requested. I haven't looked into that any further yet, may be a bug in the connector or the soapy module.

You're welcome to experiment with it though. It's a default soapy implementation, so basically all the "standard" parameters will be passed.

commit 78ccaa7d (2020-04-22): access regex groups in python 3.5 compatible way, closes #109
closed 2020-04-22 by jketterl