jketterl/openwebrx #112: DMT timeslots panel

mihaifireball, 2020-05-03:
I did I fresh install on a Ubuntu 18.04 machine and a SDRPlay RSP1 sdr, using the manual setup steps. Everything seems to work, except that the panel with the timeslots is not appearing when using DMR demodulator. The voice is getting decoded. I am using the default branch(develop). What I'm missing?
One more thing... The red warning about the code is under development can be disabled?
Thank you!

jketterl, 2020-05-03:
There was a bug in develop that prevented the DMR panel from opening. Can you try to update your checkout and see if it still applies? I've been restructuring a lot of the demodulator control code, so some bugs are expected.

The red banner is not intended to be disabled. It's remove from release versions, but as long as you're on the development branch, it will be there.

mihaifireball, 2020-05-03:
It is fixed now.
Thanks! :)

jketterl, 2020-05-03:
glad it's all good now :)

closed 2020-05-03 by jketterl