jketterl/openwebrx #114: libsoapysdr0.6 vs libsoapysdr0.7

ofadam, 2020-05-12:
For users on Ubuntu 20, how would one modify openwebrx to use libsoapysdr0.7 instead of libsoapysdr0.6?

jketterl, 2020-05-12:
Well, if you had included a little more detail I wouldn't need to guess so much.

Guessing from the question, I suppose you're attempting to install OpenWebRX from the Debian repository on an Ubuntu system. That has never been an official option (although it seems like it worked for Ubuntu 18).

I still have a solution for you: https://www.openwebrx.de/download/ubuntu.php - The owrx_connector packages in there are linked against soapy 0.7.

ofadam, 2020-05-12:
Sorry, I did indeed install from the Debian repo originally – and then upgraded to Ubuntu 20 after that, which broke soapy. I'll give this a shot – thanks!

jketterl, 2020-05-12:
May be a little tough to get out of that one because of the already-existing packages, with the same versions. It's probably best to re-install the bunch.

I didn't set up for Ubuntu 18 anymore since the LTS was imminent, and I'd probaby have run into version conflicts in the repository if I had done so. It hasn't been announced anywhere either so far, will be with the next release.

ofadam, 2020-05-12:
I just had to reinstall owrx_connector and all is well again.

jketterl, 2020-05-12:
Very well then :)

closed 2020-05-12 by jketterl