jketterl/openwebrx #115: Unable to report APRS packets

ofadam, 2020-05-13:
After filling out the appropriate APRS configuration fields, I'm able to successfully connect to the IGate server and can see openwebrx decoding APRS packets... but they don't seem to be passed along to the aprsc server in a form it will parse – it shows traffic as being sent, but no valid packets received.

I'm running DireWolf 1.5 – any suggestions on what to try?

jketterl, 2020-05-13:
Direwolf is handling all the interaction with the APRS network. There's not much I can do about that if it doesn't work, even though I'm wondering what the actual problem is. Have you tried a different gateway? How are you checking if messages are received by the gateway?

ofadam, 2020-05-13:
I've tried a few different gateways without success. Direwolf reports back which server it ends up using each time, which takes you to a status webpage for that server that shows callsigns and their traffic. Mine shows bytes being transferred, but not APRS packets.

Are there other Direwolf config flags available in the openwebrx config file beyond what's already listed? Or would I need to modify somewhere else?

jketterl, 2020-05-13:
OK, well that page should give you something indeed.

There is one specialty to consider: aprsc does not allow you to connect twice, for that reason APRS reporting has been limited to the background service only. As such, tuning to the APRS frequency in your browser will give you decodes there, but those won't be forwarded to APRS-IS.

The background service is started automatically if `services_enabled = True`, `service_decoders` contains `"packet"` and its frequency is available on the SDR, but the default config only contains the APRS frequency for Europe (144.8) which is probably not appropriate for your location. Did you edit the frequency for Packt in the `bands.json` to match the frequency in your area?

Other than that: the configuration is generated automatically, you'd need to edit the code to change it. There's just not much to be configured for an rx-only gateway apart from IGSERVER and IGLOGIN...

ofadam, 2020-05-14:
Alright, it looks like that did the trick – except that I'm having trouble with the timestamp being reported to aprs.fi. It's 1969!

https://aprs.fi/info/a/KD9NYE-10

Is there something in aprs.py that would cause an invalid timestamp somehow? Everything is fine with my server's time/date.

jketterl, 2020-05-14:
No, `aprs.py` only parses the packets for display in the frontend and use within OpenWebRX, but that data is not sent to APRS-IS. As I said, that is handled by direwolf directly.

From my knowledge of the APRS system, it might as well be you picked that time up over the air, i.e. some station local to your place that has an incorrect clock maybe. It's showing up as `1970-01-01 01:00:00 CET` for me, which is basically the equivalent of a unix timestamp of `0`.

ofadam, 2020-05-14:
Understood.

Back to the bands.json file – would it be worth adding some info about it needing to be customized for other regions? I'd be happy to help edit docs if so.

If I don't specify specific frequency settings in the main config file, does it rely on those mentioned in bands.json?

jketterl, 2020-05-14:
In the end, I'd like users to be able to chose their region on setup and be ready to go. Up until then, the configuration is basically just an intermediate solution to allow customization, albeit not with the same level of comfort. I don't have a strategy on how to get there yet, for now the `bands.json` solution works alright.

You mean not specifying profiles? No, that won't work, at least not now. I do intend to use the bandplan as an aide in the configuration process. That may also mean that the bandplan will probably see a lot more information.

ofadam, 2020-05-15:
Thanks again for all the insight - I'll close this for now.

closed 2020-05-15 by ofadam