jketterl/openwebrx #116: Digiham C compiling

IZ3GAC, 2020-05-16:
Hi All, I am trying to manual install openwebrx but compling digiham optional package, there was an issue. cmake.. command, in fact, finished his work with this info: "cmake error at cmake/FindLibMbe ... (MESSAGE) Could not find mbe".
Can anyone helm me solving? Thanks in advance.

jketterl, 2020-05-16:
There's a paragraph before digiham which takes care of mbelib, which is the part that's missing. Have you run through that?

IZ3GAC, 2020-05-16:
Many thanks, excuse my trifle...I downloaded the right folder but I missed compiling. 
Now, there is an other big issue.
The openwebrx starts correctly but: 1. there isn't the windows menu for selecting radio bands; 2. there isn't audio ... the bar graph in completely down!

jketterl, 2020-05-16:
I don't really understand the last part. I don't know what  you've done in the meantime. I don't know what's going wrong. Please check the logs!

IZ3GAC, 2020-05-16:
In the window - on the left down webpage corner - i see: "Audio underrun [0.0s]" "Audio output [0.0 Ksps]" "Audio stream [0 Kbps]" ... also squelch doesn't work. .... I try to find log and view issues. Could you give me some advices for selecting radio bands? I can choose only 70 cm and 2m by selecting "rtl-sdr usb stick 70cm relais" and "rtl-sdr usb stick 2m komplett"!

jketterl, 2020-05-16:
Yeah I'll definitely need a log to be able to tell anything.

Look, this is a place for issues, not for conversation or recommendations. Please leave it at that. Feel free to join the mailing list for smalltalk.

jketterl, 2020-05-24:
So how about that logfile? any updates? I'll close this for inactivity otherwise.

closed 2020-05-27 by jketterl