jketterl/openwebrx #117: profile for rtl_connector running on a second machine

dj1an, 2020-05-19:
Hi,
i have an rtl_connector running via systemd on a second machine.
Is it possible to define an owrx profile which connects to this daemon via ip/port?
Only solution right now seems to be soapy/soapyremote for connecting remote receivers, or am i overseeing something?

73 de DJ1AN, Jan

jketterl, 2020-05-19:
Sorry, but the owrx_connector is only intended for use by OpenWebRX running on the same machine. It was only built with a network interface for downward compatibility reasons, OpenWebRX uses a combination of nmux / nc, now you can also use owrx_connector in place of nmux. In fact, the owrx_connector is only binding it's sockets to localhost.

For now, the only available way to connect a remote device is by the means of soapy_remote source. There has been requests to add an rtl_tcp source, and this may be available in the future, but I haven't gotten around to build it yet.

jketterl, 2020-05-19:
Oh, just wanted to leave this here: If you were to connect OpenWebRX to a remote owrx_connector, you would need to run the same IQ stream over your network connection multiple times. That is part of the reason why it hasn't been designed that way. It is more efficient to have a small local process that multiplexes the source to the set of demodulators.

An OpenWebRX instance with only one connected user is already opening two IQ connections: one for the FFT (waterfall), the other for the actual audio demodulation. This increases by one for every other user.

dj1an, 2020-05-19:
Ok, thanks, that was what i expected.

closed 2020-05-19 by dj1an

jketterl, 2020-05-19:
You may want to watch this issue here: https://github.com/jketterl/openwebrx/issues/103 - that will receive all the update once I get to work on an rtl_tcp connection method.