jketterl/openwebrx #118: Confirm units of smeter pipe

jshuggins, 2020-05-21:
Hi there. Just need confirmation of the unit of the numbers coming out of the smeter pipe...

pi@openwebrx:/tmp/openwebrx $ cat openwebrx_pipe_1936400624_smeter_pipe 
1.52284e-10
1.49568e-10
1.51112e-10
1.47082e-10
1.58819e-10
1.4415e-10

Doing the math and correlating with signal strength in browser, these appear to be in linear units of mW.  True?

jketterl, 2020-05-21:
They are on a linear scale, that's all I can confirm, since the conversion to the logarithmic dB scale occurs in the JavaScript code. They aren't calibrated, so their relation to a Watt is unknown. You'd have to look at the csdr source code to find out what the actual math behind it is, but without calibrating the complete setup (including the SDR itself) this is going to be pretty much floating without much meaning.

jshuggins, 2020-05-21:
Thanks. I think I've figured it out from looking at openwebrx.js...
```
function getLogSmeterValue(value) {
    return 10 * Math.log10(value);
}

function getLinearSmeterValue(db_value) {
    return Math.pow(10, db_value / 10);
}
```
I did calibrate my AirSpy HF/OpenWebRX and your dB readings on the web display follow my signal generator dB for dB with reasonable accuracy.  That's great!  Assuming the pipe was wattage, I did the math and saw it was 30 dB off.  This suggests the unit is mW and this corresponds well with the math shown above.  So I guess the answer is the value spitting out of the smeter pipe is uncalibrated mW... or in my particular case, calibrated.

This is really very cool as I can now store (as text) and later calculate average and variance of power readings for antenna measurements against stable transmitted signals while operating via the web portal.  Very cool Jakob... thanks for continuing the work of this project.  It is more helpful than you know.

closed 2020-05-21 by jshuggins