jketterl/openwebrx #119: Please update the RPI-Images

hubertmeier, 2020-05-23:
Hi,

openwebrx is a very interesting project, there a features which a missing on the kiwiSDR (some Digimodes). The RPI-Images seems outdated, with the image from 2020-03 my PlutoSDR is not recognized, my RTL-SDR works. I see a lot of progress here.

Thank you

jketterl, 2020-05-23:
If you're talking about the images here: https://www.openwebrx.de/download/rpi.php

They are up to date. There hasn't been any releases since then. There hasn't been any work on the plutosdr either since then, so if there's something wrong with it, I suppose a simple update of the images won't help.

hubertmeier, 2020-05-24:
ok, thanks, i thougt they were outdated because i saw a lot of commits since march ...  can i update the project on the pi? First i tried a full install on the pi but there was an error with make at the first steps ... So I took the image again.

thank you for your work

jketterl, 2020-05-24:
As I said, there hasn't been any releases since 0.18.0. The fact that there's a lot of commits means there is a lot of development going on, which may eventually lead up to the next release, once everything is stable and confirmed working.

The images are based on the available repositories and packages, where available, so you will be able to grab updates from there once a new release becomes available.

There is an experimental repository available which contains automatically built packages that become available pretty much minutes after new commits are pushed here. However, this is *not* intended for production use. It is available to test new features and possible support development. I will assume a certain level of experience with system administration if you use it.

https://www.openwebrx.de/download/debian.php

If you only came here to enjoy the new features, the message is: Wait for the next release.

closed 2020-05-27 by jketterl