jketterl/openwebrx #120: Unable to install CSDR

ofadam, 2020-05-25:
Any idea why these commands: 
`git clone https://github.com/jketterl/csdr.git
cd csdr
make
sudo make install
cd ..`

Results in this error:
`make: *** No targets specified and no makefile found. Stop.`

I installed all prerequisites on Ubuntu 20.04.

jketterl, 2020-05-25:
Oh. yes, the build process changed, it's now automake-based. The readme on csdr has the right instructions. I thought I had updated the wiki, but it turns out all I did was add the required packages. Sorry about that, it's updated now.

ofadam, 2020-05-25:
That did the trick – if upgrading CSDR on an existing install, do I need to do anything but a git pull?

jketterl, 2020-05-25:
I'd say it depends on the distribution. The new system finally installs the binaries correctly, so they should end up in `/usr/local` in most cases. On most systems, that takes precedence over the old locations, but I still suggest to manually clean the existing files to make sure:

```
/usr/bin/csdr
/usr/bin/csdr-fm
/usr/bin/nmux
/usr/lib/libcsdr*
/usr/include/libcsdr.h
```
(list is from memory. done this on a few boxes by now...)

Please also note that I put an additional `ldconfig` in the wiki now, helps with first installation.

jketterl, 2020-06-01:
Any updates on this? Did you succeed?

ofadam, 2020-06-03:
Yep, this seems to have worked. Thanks!

closed 2020-06-03 by ofadam