jketterl/openwebrx #126: FreeDV Codec 2 support

mkopa, 2020-06-01:
Is it possible to add FreeDV (Codec 2) decoder support? It would be awesome to have a digital voice on low bandwidth HF bands in a fantastic OpenWebRX project.

http://www.rowetel.com/?page_id=452
https://github.com/drowe67/codec2

jketterl, 2020-06-01:
This has been on the ToDo list pretty much from the beginning. Unfortunately, I haven't found any decoder software that would be able to process this on the commandline. Anything that runs in OpenWebRX must be able to run without GUI. To save space, I'd even prefer if it didn't have a GUI to begin with.

eroyee, 2020-06-02:
Greetings,

I've never used FreeDV (but have been aware of it for some years) however I believe it has an API.

From my brief reading there appear to be some examples that use piping via the cli, and which don't use the GUI. It's not completely clear to me that the GUI is necessary.

Good work on OpenwebRX BTW,  thanks!

eroyee, 2020-06-03:
Just to (hopefully) provide more useful comment .. the following worked for me in Ubuntu 18.04:

git clone https://github.com/drowe67/codec2.git
cd codec2
mkdir build
cd build
cmake ..
make

I then ran the following on one of the provided sample .wav files:

./src/freedv_tx 700C ../wav/all.wav - | ./src/freedv_rx 700C - - | aplay -f S16_LE -r 8000

Result was, as expected, quite intelligible but clearly compressed audio out (whereas the input .wav file sounded 'normal'), and the following at completion:

Playing raw data 'stdin' : Signed 16 bit Little Endian, Rate 8000 Hz, Mono frames decoded: 713  output speech samples: 456320

I also tried with a higher bit rate:

./src/freedv_tx 1600 ../wav/all.wav - | ./src/freedv_rx 1600 - - | aplay -f S16_LE -r 8000

With similar audible results, and following out to screen:

Playing raw data 'stdin' : Signed 16 bit Little Endian, Rate 8000 Hz, Mono frames decoded: 2857  output speech samples: 453120

A further check of David Rowe's repositories suggests that the FreeDV GUI is separate to the command-line tools.

HTH.

mkopa, 2020-06-03:
On the docker image - jketterl/openwebrx works fine too.

// start container 
docker run --device /dev/bus/usb -p 8073:8073 -v openwebrx-config:/etc/openwebrx --name openwebrx jketterl/openwebrx:stable

// connect to bash container
docker exec -it openwebrx bash

apk update
apk add --update-cache --no-cache git cmake g++ build-base
git clone https://github.com/drowe67/codec2.git
cd codec2
mkdir build
cd build
cmake ..
make

jketterl, 2020-06-03:
alright, this may be turning into a low-hanging fruit then :slightly_smiling_face: will need to give this a whirl.

jketterl, 2020-08-29:
Alright, this seems to be getting somewhere. I just did some agc fine-tuning, but it's hard to judge if this is working alright since I'm struggling to find *any* signal. I have done most testing using my own transmitter, which is probably not the best reference since I haven't used that to do any actual QSOs.

The mode requires manual installation of the freedv_rx binary, it is not included with any packages, nor is it installed with `sudo make install` when compiling. This is probably a big drawback, I can already see a lot of people requesting instructions on how to set this up. I can fix this for the RPi and docker images.

closed 2020-11-04 by jketterl

darksidelemm, 2020-12-31:
Probably worth pointing out that FreeDV mode 700D is probably as popular as mode 1600 now, due to it's significant robustness. All the modes are decodable from freedv_rx, and take the same kind of input (SSB audio), so being able to select between the available modes would be useful :-)  (1600, 700C, 700D, and recently 700E).