jketterl/openwebrx #127: Support for AFEDRI SDRs (sdr-net, Lan IQ SDR)

dm6tt, 2020-06-01:
Hi Jakob,
I really appreciate that you took over the development of openwebsdr.
I got the last version from HA7ILM running with a sdr-net device from 4Z5LV (AFEDRI-SDRs). It is described in the github.

I would also appreciate if you would add support for the AFEDRI devices, since this low cost network SDRs are a great choice.
You only need two clis (sdr_commander, sdr_split).
Links:
https://github.com/ha7ilm/openwebrx/wiki/Using-OpenWebRX-with-AFEDRI-SDR
https://www.afedri-sdr.com/index.php/downloads/category/15-sdr-commander-command-shell-utility
https://www.afedri-sdr.com/index.php/downloads/category/24-source-code-directory

Thanks
73, Marcus, DM6TT

jketterl, 2020-06-01:
oh this is terrible. who comes up with these apis? two separate command line tools to control one SDR, and they both need to be fed with the sample rate and frequencies? how will such a device handle a frequency change?

this will have to wait, simply because i cannot implement this blindly. in the meantime, you can contact the manufacturer and ask them if they wouldn't want to come up with a soapysdr module for it.

ch0mik, 2024-01-15:
https://github.com/jketterl/openwebrx/blob/develop/docker.sh#L5C78-L5C94 ?

jketterl, 2024-01-15:
Oh, indeed. Afedri is now available in the latest development version. Forgot about this issue here.

Credit goes to @alexander-sholohov who implemented the SoapySDR module and provided a pull request for OpenWebRX.