jketterl/openwebrx #128: Warning message - problem with Debian repository?

DL5SMB, 2020-06-02:
After I crossgraded my system from i386 to amd64 architecture, I assumed that the previously compiled source-code version would not work anymore.

However as the repository contains amd64-versions I decided using the repos was most probably easier for updating on further releases in future. So I deleted the source-code and switched to the repo. However I noticed I received a small, probably unimportant warning after installing via the repo:

```
`openwebrx (0.18.0) wird eingerichtet ...
**+ adduser --system --group --no-create-home --home /nonexistant openwebrx** // is this correct?
Warnung: Auf das von Ihnen angegebene Home-Verzeichnis /nonexistant kann nicht zugegriffen werden: No such file or directory
Lege Systembenutzer »openwebrx« (UID 131) an ...
Lege neue Gruppe »openwebrx« (GID 147) an ...
Lege neuen Benutzer »openwebrx« (UID 131) mit Gruppe »openwebrx« an ...
Erstelle Home-Verzeichnis »/nonexistant« nicht.
```
`
I do not know if this is caused by the previous manual installation or by a mistake in the installation files in the repo - I just wanted to share in case it is caused by the latter.

jketterl, 2020-06-02:
It's not really an issue, at least to my knowledge this shouldn't cause any problems. The intention of the installation is not to create a home directory for the user, which is basically what's happening.

The "home" directory is a mandatory field in the Linux user management, setting it to `/nonexistant` will give whoever comes across this user in the future a clear information that this user does not own a home directory intentionally. This is a pattern that seems to be used for other users in the Debian environment, too (e.g. for the "nobody" user), even though there is no official documentation of this.

DL5SMB, 2020-06-02:
Okay then, thanks for having a look. 
I had not noticed any further problems or repercussions, but better safe then sorry.

closed 2020-06-02 by DL5SMB