jketterl/openwebrx #130: Version 19 /tmp/openwebrx

jshuggins, 2020-06-03:
Interesting new files in the tmp directory...

pi@openwebrx:/tmp/openwebrx $ ls -l
total 12
-rw-r--r-- 1 openwebrx openwebrx 71 Jun  3 00:30 '\b'\'''$'\200''~dE'$'\363''v'$'\001'
-rw-r--r-- 1 openwebrx openwebrx 71 Jun  3 00:28 ''$'\b\307\333''~du'$'\370''v'$'\001'
-rw-r--r-- 1 openwebrx openwebrx 71 Jun  3 00:30 ''$'\b\a\237''~d'$'\325\366''v'$'\001'
prw-r--r-- 1 openwebrx openwebrx  0 Jun  3 00:34  openwebrx_pipe_1969567472_bpf_pipe
prw-r--r-- 1 openwebrx openwebrx  0 Jun  3 00:34  openwebrx_pipe_1969567472_shift_pipe
prw-r--r-- 1 openwebrx openwebrx  0 Jun  3 00:39  openwebrx_pipe_1969567472_smeter_pipe
prw-r--r-- 1 openwebrx openwebrx  0 Jun  3 00:31  openwebrx_pipe_1969567472_squelch_pipe
pi@openwebrx:/tmp/openwebrx $

The pipes are find, but wonder what the first three represent.

jketterl, 2020-06-03:
They are coming from wsjt-x... It's actually simple to reproduce, enter an empty directory and run `jt9` without any argument, you'll have one of those files afterwards.

I have had a look at the sources, the problem is that it simply writes to the file using an unintialized variable as its name, so this is probably just some random garbage from memory. I haven't attempted to fix it since I have no idea on where to send the bugfix. It would probably also be a good idea to check if this is still present in the upcoming 2.2 release.

The files only appear when a feature check is performed by OpenWebRX, testing the availability of the `jt9` command is part of that. I don't have a complete list of when those checks run, but it's on-demand and uncached, so there will probably be more of these files with time.

There may be ways to work around the problem, I haven't attempted to figure out a specific set of arguments that may avoid the bug. Introducing a feature result cache may lower the number of files produced.

jshuggins, 2020-06-03:
I did try running jt9 with no arguments in an empty directory.  It responds with a command list, but no file made. Regardless this seems harmless enough and is something I've not seen until the 0.19 R-Pi image that I'm running now.

jketterl, 2020-06-03:
That's odd. I just tried the same and ended up with one such file, as expected.

The symptoms themselves have been there previously, but the configuration option was not passed correctly in the last release, so those files were ending up in `/tmp` before.

RipperFox, 2020-06-11:
It seems it's the 'js8' command, which creates the files, not 'jt9'.

Woth to mention is that the default temp directory has been changed from /tmp/openwebrx to '/tmp'. Only /tmp/openwebrx was mounted as tmpfs (=in RAM) in my RasPi install, so I had to change my fstab after the upgrade to 0.19.

jketterl, 2020-06-11:
It's not suprising to see js8 affected as well, the decoder is basically using the same codebase, at least to my knowledge.

RipperFox, 2020-06-11:
I think js8 and jt9 share the same code which creates "empty" wisdom files even when just the help is printed. But in jt9 filename is empty, so no file is actually created. In js8 I suspect the filename is just random garbage from an uninitialized variable. Maybe they use different compiler options?

See also my comments here: https://bitbucket.org/widefido/js8call/issues/342/rasbian-arm-linux-js8-creates-unnecessary

jketterl, 2020-06-11:
I've been looking at the WSJT-X (2.1.2) source code, the problem with the unintialized variable stems from there, at least if I sufficiently understood it. I'm no Fortran expert.

There's a block in `lib/jt9.f90` that outputs the usage message, closing off with a `go to 999` in line 158, which goes pretty much straigh to the end. After 999 though, there's the call to `fftw_export_wisdom_to_filename(wisfile)`. That variable `wisfile` is define on line 21, but its value is assigned on line 167, after the jump to 999 when the usage is output, so it doesn't receive a value and most likely contains memory garbage in this particular case.

RipperFox, 2020-06-11:
I'm also no Fortran expert.. but that's exactly my conclusion in the mentioned bug report there :)

jketterl, 2020-07-21:
I just had a look at WSJT-X 2.2.2, I don't see it fixed in the code. JS8Call 2.2.0 doesn't look much better from testing...

jketterl, 2020-07-25:
Alright, here's a patch against WSJT-X 2.2.2. The easiest way to apply this is to put it into the `wsjtx.patch` file that comes inside the tarball before compiling. I will see if I can get some attention on the mailing list. I'm not too sure how this will work, I couldn't locate any issue tracker for WSJT-X.

```
diff -ur wsjtx-orig/lib/jt9.f90 wsjtx/lib/jt9.f90
--- wsjtx-orig/lib/jt9.f90      2020-07-21 21:10:43.561474123 +0200
+++ wsjtx/lib/jt9.f90   2020-07-25 20:39:36.413518153 +0200
@@ -77,6 +77,7 @@
   data npatience/1/,nthreads/1/
 
   nsubmode = 0
+  wisfile=C_NULL_CHAR
 
   do
      call getopt('hs:e:a:b:r:m:p:d:f:w:t:98654qTL:S:H:c:G:x:g:X:',      &
@@ -340,7 +341,7 @@
 !            58('-')/6i8)
 
 ! Save wisdom and free memory
-  iret=fftwf_export_wisdom_to_filename(wisfile)
+  if (len(wisfile) > 0) iret=fftwf_export_wisdom_to_filename(wisfile)
   call four2a(a,-1,1,1,1)
   call filbig(a,-1,1,0.0,0,0,0,0,0)        !used for FFT plans
   call fftwf_cleanup_threads()
```

Also, this patch should apply to js8call as well. Their codebase stems from WSJT-X, and the `js8` command line decoder sources reside under the exact same file`lib/jt9.f90`. The only difference is that you'll have to apply it to the sources directly.

jketterl, 2020-07-27:
Good news: I've received a response from Joe Taylor himself saying that a fix should be included with the next release.

I hope that this will propagate to JS8Call too, eventually, or maybe on request. We'll see.

Apart from that, my testing setup seems to be pretty well-behaved by now. The only scenario where I see leftovers is when a forced shutdown occurs. It may be worth having another look at leaking threads (they can potentially block the shutdown), but during normal operation I don't see any leftovers.

closed 2020-08-26 by jketterl

jketterl, 2020-12-08:
Good news for anybody still waiting for an update here: I have checked the new version 2.3.0rc2 (release candidate) of WSJT-X, and there's now a check in place that prevents those weird-named temporary files from being written. Verified in practice, too.

JS8 is still affected, so this will still need some more time to be completely eradicated.