jketterl/openwebrx #131: In 0.19, scheduling no longer works

jshuggins, 2020-06-03:
Perhaps the 0.19 version has changed the syntax of the scheduling in the config file.  This...

        "scheduler": {
            "type": "daylight",
            "schedule": {
                "night": "40m",
                "day": "30m",
                "greyline": "20m",
            },
        },

...worked with the March version very well, but doesn't seem to do anything in 0.19.  The log files do reveal this one exception when I was messing around with the web interface last night...

Jun  3 01:43:05 openwebrx openwebrx[2098]: 2020-06-03 01:43:05,849 - owrx.websocket - DEBUG - websocket close frame received; closing connection
Jun  3 01:43:05 openwebrx openwebrx[2098]: 2020-06-03 01:43:05,849 - owrx.websocket - DEBUG - websocket loop ended; shutting down
Jun  3 01:43:05 openwebrx openwebrx[2098]: closing client socket
Jun  3 01:43:05 openwebrx openwebrx[2098]: 2020-06-03 01:43:05,862 - csdr.csdr - DEBUG - dsp thread ended with rc=-15
Jun  3 01:43:05 openwebrx openwebrx[2098]: Exception in thread Thread-106:
Jun  3 01:43:05 openwebrx openwebrx[2098]: Traceback (most recent call last):
Jun  3 01:43:05 openwebrx openwebrx[2098]:   File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
Jun  3 01:43:05 openwebrx openwebrx[2098]:     self.run()
Jun  3 01:43:05 openwebrx openwebrx[2098]:   File "/usr/lib/python3.7/threading.py", line 865, in run
Jun  3 01:43:05 openwebrx openwebrx[2098]:     self._target(*self._args, **self._kwargs)
Jun  3 01:43:05 openwebrx openwebrx[2098]:   File "/usr/lib/python3/dist-packages/owrx/connection.py", line 37, in mp_passthru
Jun  3 01:43:05 openwebrx openwebrx[2098]:     self.send(data)
Jun  3 01:43:05 openwebrx openwebrx[2098]:   File "/usr/lib/python3/dist-packages/owrx/connection.py", line 44, in send
Jun  3 01:43:05 openwebrx openwebrx[2098]:     self.conn.send(data)
Jun  3 01:43:05 openwebrx openwebrx[2098]:   File "/usr/lib/python3/dist-packages/owrx/websocket.py", line 105, in send
Jun  3 01:43:05 openwebrx openwebrx[2098]:     raise WebSocketClosed()
Jun  3 01:43:05 openwebrx openwebrx[2098]: owrx.websocket.WebSocketClosed
Jun  3 01:43:05 openwebrx openwebrx[2098]: 2020-06-03 01:43:05,864 - owrx.websocket - DEBUG - websocket loop ended; sending close frame
Jun  3 01:43:20 openwebrx openwebrx[2098]: signal 15 caught
Jun  3 01:43:20 openwebrx openwebrx[2098]: closing client socket
Jun  3 01:43:20 openwebrx openwebrx[2098]: closing client socket

The log file has been absolutely quiet concerning openwebex for just over twelve hours until I just now accessed via the web interface again.  The band that came up is my lowest frequency rather than what is scheduled.

Will mess with this some more.

jketterl, 2020-06-03:
No changes to the syntax. Exception indicates a client has disconnected, not much more.

jshuggins, 2020-06-03:
I'll move this 0.19 version to my more experimental setup and change my main system back to its earlier state.

jshuggins, 2020-06-03:
During migration of config files, I forgot to set services_enabled to true.  Thinks look promising at the moment and will report back in a few minutes.

closed 2020-06-03 by jshuggins