jketterl/openwebrx #136: Tuning Step Selector (feature request)

frohro, 2020-06-22:
Hi Jacob,
Thanks for all your great work on openwebrx.  I played with it a bit today, and one thing i miss when tuning SSB and AM is a way to click and get the frequency right.  These days, most transmitters have digital frequency adjustment, and on other software I have had the ability to select the tuning step, so it will end on the frequency I click on, rounded to the tuning step.  For example, for SSB, I like a tuning step of 1 kHz.  When I click on 14.275200 MHz, it would round it to 14.275000 MHz.  That would be a nice feature for openwebrx.
73,
Rob
KL7NA

jketterl, 2020-06-22:
Have you tried using the scroll wheel on the digits of the frequency? That method should give you steps of 100Hz, 1kHz, 10kHz and so on.

frohro, 2020-06-22:
Hi Jakob,
I had not tried that.  I just did, but it doesn't do what I want; you have to edit the frequency box to get it to an "even" frequency.  It is a nice way to emulate the old fashioned tuning dial though.  If I want to tune in a signal I see with the scroll wheel after I set the remaining frequency part to 0000, then it works, but because my scroll wheel is a bit too sensitive, I really would prefer to be able to click on the spectrum.  I think it might be nice to be able to click on a digit and it would highlight one of the 10 kHz, 1 kHz, or 0.1 kHz digits, making all the less significant digits 0, and then when clicking on a frequency it will round to the closest "even" frequency.  
73,
Rob
KL7NA

jketterl, 2020-06-22:
Yeah, I see. The problem with your proposed request is that it doesn't translate well to all bands or modes (in FM the grid is usually 12,5kHz), and you'd probably not want to have it on permamently. I'm also reluctant to add a bunch of new controls that are hard to understand for new users.

frohro, 2020-06-22:
Yes, you are right about the unexplained controls.  I hadn't thought about the 12.5kHz steps.  Maybe, a  tuning step drop down menu would be better.

73,
Rob
KL7NA