jketterl/openwebrx #137: High CPU load on client machine

eroyee, 2020-06-23:
Noting below for reference. Using OpenwebRX dev version from ~7 days ago.

In use the receiver and status overlays put a significant load on a client machine.

Switching off both of them reduces the load, switching off one or the other makes little difference.

**Background:**

I was doing some testing on network bandwidth requirements because I have a limited uplink connection. I used a middle-aged Celeron laptop running latest Mint 19.3 as the psuedo-client.

The laptop battery seemed to go flat more quickly than usual. Top showed constant CPU use of +/-80% when OpenwebRX was in use.

Switching off receiver AND status overlays reduced CPU to < 40%.

This is replicable to an i7 client, the increase in load on that is in the region of 10-15%.

I've not looked into it further but guess this is generated by the dynamic components of the overlays? 

Out of interest I tested a KiwiSDR instance and it didn't load the CPU the same (with the full overlay on). CPU use was similar to an OpenwebRX instance with both overlays turned off.

I will see if I can narrow down what component(s) of the overlay cause this and will report back if so.

jketterl, 2020-06-23:
This is probably down to the animations. OpenWebRX uses both CSS and JavaScript animations in the frontend. I'm afraid there's nothing I can do about that, short of removing them (which is not an option, IMO). In the end, it's not OpenWebRX itself that is consuming the CPU, but the browser and it's rendering engine. I supposee the results will vary across different browsers and versions. Also, in my experience, Linux-based systems are expecially inefficient at drawing graphics, probably because they're not making full use of the hardware acceleration.

eroyee, 2020-06-23:
Thanks Jakob,

I should have mentioned the browser - FF 77.0.1. 

I agree it likely involves the animations, but I was interested to see that the increase in load didn't occur with KiwiSDR with its S-meter bar in operation. I've mentioned before than I know nothing about js but I did see a couple of mathematical functions for the signal-bar in OpenwebRX and wondered if that processing might have had an affect?

To my mind the status panel doesn't matter (ie. it's not central to functioning, and it could default to off) but of course the receiver panel and signal-bar are needed. I'll try a different browser tomorrow and will also do a bit more general research around this. If anything useful arises I'll let you know.

jketterl, 2020-06-23:
I've just had a look, and it doesn't look as if the S-Meter in KiwiSDR still had anything to do with what was / is implemented in OpenWebRX. The implementation there is using a canvas object where you can basically draw your own graphics. It's probably a lot more code, and as you'd expect from such a solution, the graphical appeal is very low, kind of like 90s software before Win95 came along. I guess that's the price you pay if you optimize for performance.

As I said before, I'm afraid there's little I can do about this. The status panel can already disabled, and one day I may get around to implement some kind of personal state that's stored in your browser that would keep it off once it's disabled.

I'll close this since there's nothing really broken. The mailing list is available for further discussion or questions.

closed 2020-06-23 by jketterl