jketterl/openwebrx #138: Registration site

Challado, 2020-06-23:
Dear friends.
While sdr.ru stops working / registrating new members to put the radio listen to anyone, how are the substitute / workaround to put my new openwebrx device disponible to anything? Exists a substitute or another way to do that?

jketterl, 2020-06-23:
There's a separate project that I have started. Please checkout https://www.receiverbook.de/

It's just a few weeks old, so I guess it will take some time until it makes its rounds...

Challado, 2020-06-23:
Ah, ok.
Doesn't have anything to put on config_webrx.py to register / unregister to
see the SDR status online / offline for now?

jketterl, 2020-06-23:
Not necessarily, but that depends on the exact scenario. You may need to add a `receiver_key`. But the site should guide you through the process.

Challado, 2020-06-23:
Ok. When I will put the SDR online I'll back to site. The registration process is OK. I'll go when I have a valid ip address to do so.

closed 2020-07-07 by jketterl