jketterl/openwebrx #141: Numbers are jumping too fast

T-Shilov, 2020-06-26:
Dear Jakob,

When working with the SQ meter, I found that the numbers are jumping too fast, they cannot be remembered (in attacment).

[Numbers are jumping too fast.avi.zip](https://github.com/jketterl/openwebrx/files/4834837/Numbers.are.jumping.too.fast.avi.zip)

This especially interferes with the measurement of ether noise, and these measurements are also important.
Please remove the 3rd digit in the SQ meter. Or integrate them over a longer period of time.
The radio ether is a very volatile и unstable environment, and it makes no sense to measure the he level accurate to the 3th sign (0.1%).

Best regards,
T. Shilov
UR1229SWL

jketterl, 2020-06-26:
I don't think you should be taking important measurements using OpenWebRX. There is no calibration, and the signal meter is tied to quite a bunch of variables.

T-Shilov, 2020-06-26:
Dear Jakob, you contradict yourself :-)
If SK is so inaccurate, why do need the 3rd digit?

jketterl, 2020-06-26:
Am I? Please double-check your assumptions. This meter (including its 3 digits) has been in place since before I took over the project.

Your subversive style of discussing things is putting yet another feature at risk. If you're really going to question the number display as it is, I may as well remove it, right? Watch where you're going with your arguments.

T-Shilov, 2020-06-26:
> This meter (including its 3 digits) has been in place since before I took over the project.

Right. 
However, no one reproaches you that the previous project had such an unsuccessful 3-digit solution.
But now you're running the project now, aren't you? 
Why don't you make this improvement in it?

>  If you're really going to question the number display as it is, I may as well remove it, right? 

Right. You can do this stuff.
This really it is  your the subversive and aggressive style of discussing things instead of constructive discussion.

How can you not understand that with 3 numbers jumping like a flea, me have to wait a long time to see the minimum signal level, then wait a long time for the maximum signal level.
Then fold  them up and divide them in half to get the average value of the signal.
Therefore, me have to stare at the screen for a long time, and this spoils my vision.
Do you want users OpenWebRX to lose their eyesight??

If you do not want to remove the 3rd digit, I have already offered you the second option:  just do the integration over a longer time interval so that the numbers change less frequently.

jketterl, 2020-06-26:
> However, no one reproaches you that the previous project had such an unsuccessful 3-digit solution.

You just did.

And now you're even implying that it's unsuccessful. Just stop judging everything, will you?

> Why don't you make this improvement in it?

I have done so and I will continue to do so, one at a time.

> How can you not understand that with 3 numbers jumping like a flea, me have to wait a long time to see the minimum signal level, then wait a long time for the maximum signal level.

At no point have I questioned the actual idea, or your suggestions. I have added the "feature" flag and it's now on the backlog.

All I did was pointing out that you may be using this feature in a way that it was not designed, and that you may still be disappointed even with the improvement in place.

T-Shilov, 2020-06-26:
Jakob, I want you to understand correctly: my proposals are in the nature of petition only, and nothing more.
Perhaps the language barrier prevents hinders you from understanding this, because I use a Google translator, but it doesn’t work very well.

By the way, a colleague [Excalibur201010 ](https://github.com/jketterl/receiverbook/issues/2)

already done this, he just slowed down the speed of the change of numbers - http://sdr.mis.ks.ua:9000/

But I’d better wait for your formal decision.