jketterl/openwebrx #142: Broken Pipe

dg9vh, 2020-06-26:
Hi,

I get this kind of error once/twice a day and in fact of this I have to restart the service. What is the cause of this and how could this be fixed?

```
Jun 26 07:36:53 websdr openwebrx[22161]: 2020-06-26 07:36:53,497 - owrx.websocket - DEBUG - websocket close frame received; closing connection

Jun 26 07:36:53 websdr openwebrx[22161]: 2020-06-26 07:36:53,498 - owrx.websocket - DEBUG - websocket loop ended; shutting down

Jun 26 07:36:53 websdr openwebrx[22161]: 2020-06-26 07:36:53,502 - csdr.csdr - DEBUG - dsp thread ended with rc=-15

Jun 26 07:36:53 websdr openwebrx[22161]: signal 15 caught

Jun 26 07:36:53 websdr openwebrx[22161]: 2020-06-26 07:36:53,506 - csdr.csdr - DEBUG - dsp thread ended with rc=-15

Jun 26 07:36:53 websdr openwebrx[22161]: 2020-06-26 07:36:53,510 - owrx.source - DEBUG - shut down with RC=0

Jun 26 07:36:53 websdr openwebrx[22161]: 2020-06-26 07:36:53,511 - owrx.cpu - DEBUG - cpu usage thread shut down

Jun 26 07:36:53 websdr openwebrx[22161]: 2020-06-26 07:36:53,512 - owrx.websocket - DEBUG - websocket loop ended; sending close frame

Jun 26 07:36:53 websdr openwebrx[22161]: Traceback (most recent call last):

Jun 26 07:36:53 websdr openwebrx[22161]:   File "/usr/lib/python3.7/multiprocessing/queues.py", line 242, in _feed

Jun 26 07:36:53 websdr openwebrx[22161]:     send_bytes(obj)

Jun 26 07:36:53 websdr openwebrx[22161]:   File "/usr/lib/python3.7/multiprocessing/connection.py", line 200, in send_bytes

Jun 26 07:36:53 websdr openwebrx[22161]:     self._send_bytes(m[offset:offset + size])

Jun 26 07:36:53 websdr openwebrx[22161]:   File "/usr/lib/python3.7/multiprocessing/connection.py", line 404, in _send_bytes

Jun 26 07:36:53 websdr openwebrx[22161]:     self._send(header + buf)

Jun 26 07:36:53 websdr openwebrx[22161]:   File "/usr/lib/python3.7/multiprocessing/connection.py", line 368, in _send

Jun 26 07:36:53 websdr openwebrx[22161]:     n = write(self._handle, buf)

Jun 26 07:36:53 websdr openwebrx[22161]: BrokenPipeError: [Errno 32] Broken pipe

Jun 26 07:36:53 websdr openwebrx[22161]: Traceback (most recent call last):

Jun 26 07:36:53 websdr openwebrx[22161]:   File "/usr/lib/python3.7/multiprocessing/queues.py", line 242, in _feed

Jun 26 07:36:53 websdr openwebrx[22161]:     send_bytes(obj)

Jun 26 07:36:53 websdr openwebrx[22161]:   File "/usr/lib/python3.7/multiprocessing/connection.py", line 200, in send_bytes

Jun 26 07:36:53 websdr openwebrx[22161]:     self._send_bytes(m[offset:offset + size])

Jun 26 07:36:53 websdr openwebrx[22161]:   File "/usr/lib/python3.7/multiprocessing/connection.py", line 404, in _send_bytes

Jun 26 07:36:53 websdr openwebrx[22161]:     self._send(header + buf)

Jun 26 07:36:53 websdr openwebrx[22161]:   File "/usr/lib/python3.7/multiprocessing/connection.py", line 368, in _send

Jun 26 07:36:53 websdr openwebrx[22161]:     n = write(self._handle, buf)

Jun 26 07:36:53 websdr openwebrx[22161]: BrokenPipeError: [Errno 32] Broken pipe

Jun 26 07:36:53 websdr openwebrx[22161]: Traceback (most recent call last):

Jun 26 07:36:53 websdr openwebrx[22161]:   File "/usr/lib/python3.7/multiprocessing/queues.py", line 242, in _feed

Jun 26 07:36:53 websdr openwebrx[22161]:     send_bytes(obj)

Jun 26 07:36:53 websdr openwebrx[22161]:   File "/usr/lib/python3.7/multiprocessing/connection.py", line 200, in send_bytes

Jun 26 07:36:53 websdr openwebrx[22161]:     self._send_bytes(m[offset:offset + size])
[... 822 lines cut ...]

Jun 26 07:36:53 websdr openwebrx[22161]:   File "/usr/lib/python3/dist-packages/owrx/connection.py", line 44, in send

Jun 26 07:36:53 websdr openwebrx[22161]:     self.conn.send(data)

Jun 26 07:36:53 websdr openwebrx[22161]:   File "/usr/lib/python3/dist-packages/owrx/websocket.py", line 105, in send

Jun 26 07:36:53 websdr openwebrx[22161]:     raise WebSocketClosed()

Jun 26 07:36:53 websdr openwebrx[22161]: owrx.websocket.WebSocketClosed

Jun 26 07:36:53 websdr openwebrx[22161]: 2020-06-26 07:36:53,600 - owrx.websocket - ERROR - OSError while writing data

Jun 26 07:36:53 websdr openwebrx[22161]: Traceback (most recent call last):

Jun 26 07:36:53 websdr openwebrx[22161]:   File "/usr/lib/python3/dist-packages/owrx/websocket.py", line 133, in _sendBytes

Jun 26 07:36:53 websdr openwebrx[22161]:     written = self.handler.wfile.write(chunk)

Jun 26 07:36:53 websdr openwebrx[22161]:   File "/usr/lib/python3.7/socketserver.py", line 799, in write

Jun 26 07:36:53 websdr openwebrx[22161]:     self._sock.sendall(b)

Jun 26 07:36:53 websdr openwebrx[22161]: BrokenPipeError: [Errno 32] Broken pipe
```

jketterl, 2020-06-26:
This happens because the shutdown of the csdr processes that deliver your FFT and audio data is not synchronized. They still produce data for some time after they are killed, but the data cannot be sent to the client because the socket has been closed (`websocket close frame received; closing connection`).

This has been uncoupled using queues and threads, and although some threads will probably be crashing due to this, they should all be tied to that specific client and thus do no damage. I cannot see any evidence of anything else in this log, and you should not need to restart the service because of this.

dg9vh, 2020-06-26:
I need to restart, because following connections get an error-message, that no SDR-receiver is available. If I could detect this with an watchdog-script, I could restart based on this... You say that csdr process would be killed or something... I would investigate this, if this is a point to catch and restart automatically...

jketterl, 2020-06-26:
I don't see any messages indicating an SDR device failure (there's usually messages by the DSP and FFT modules responding to the failure). Please see if the device fails during connection of the second client.

dg9vh, 2020-06-26:
I think the message happend when connecting with a second client... would observe this... we have normally one to two clients here in the house .-) so this would be reproduceable.

jketterl, 2020-07-07:
has been resolved on the side. caused by invalid sample rate.

closed 2020-07-07 by jketterl