jketterl/openwebrx #143: Prepare software for hardware upgrade help

cTurtle98, 2020-07-02:
I downloaded the raspberry pi image version of openwebrx and put it on an SD card and into a pi 3b 

I am running out of cpu so I purchased a pi4 b 2gb

What do I need to do to prepare the os on the sd card so when the new pi gets here I can just pop out the sd card and put it in the new one?

jketterl, 2020-07-02:
The only thing to consider is when you downloaded the image. Anything that has been released since ~January came with the Raspberry Pi 4 support, so you should be able to just swap them out. If your image is older than that, I'd recommend starting from scratch since you're running a pretty outdated version and don't have any update support. All of those issues were resolved by the time of the OpenWebRX 0.18.0 release, so it's very likely you're not affected.

closed 2020-07-11 by jketterl

cTurtle98, 2020-07-15:
I got my pi 4 and swapped the sd card in and everything is working as far as I can tell. thank you