jketterl/openwebrx #146: OpenWebRX suddenly stopped working

T-Shilov, 2020-07-07:
A very unpleasant event :(

OpenWebRX worked well. But a few days ago, he suddenly stopped his work, and I did not immediately notice this.
I do not understand what caused this.

I am using OpenWebRX v0.19.1 and Debian 10/64 bit.

The error log is in the attachment.
[daemon.log.tar.gz](https://github.com/jketterl/openwebrx/files/4883372/daemon.log.tar.gz)

jketterl, 2020-07-07:
The logfile indicates that you have hit the "too many files" problem, that I'm also seeing on one of our receivers. It happens after an uptime of some weeks, but I haven't had the time to dig into what is exactly going on. Initial analysis suggests that the pipes in the csdr demodulators are not managed correctly and their file descriptors are leaking.

T-Shilov, 2020-07-07:
Ok, I understand you, Yakob. I will wait for your solution to this problem.

T-Shilov, 2020-07-08:
**Scheisse, today this problem repeated itself!**
But I can’t constantly monitor the server and restart OpenWebRX!

Jakob, when will you solve this vile problem?

Excalibur201010, 2020-07-10:
Good evening, i have some issue, after 2-3 days working OpenWebRX cant accept new connections. Maximum peoples listening thr radio is 5-6.
Thanks and best regards US1GHQ

jketterl, 2020-07-11:
@Excalibur201010 does your log show the same "too many open files" messages?

Excalibur201010, 2020-07-12:
Yes in some message i seen this line, but now working 3days stabilng, if owrx stoping again i post my log.

jketterl, 2020-07-12:
That'll do for now. just wanted to make sure it's the same issue.

Excalibur201010, 2020-07-15:
Good day finaly OpenwebRX i stuck after 7 days working. Now a put my logs when is started.
After restart service, not OS, OWRX working fine

`
Jul 15 10:35:50 rds-2 LINBPQ[14620]: BPQ32 Heartbeat Buffers 981#015
Jul 15 10:36:27 rds-2 openwebrx[3262]: SSSSSSSS2020-07-15 10:36:27,166 - owrx.connection - DEBUG - client connection intitialized
Jul 15 10:36:28 rds-2 openwebrx[3262]: 2020-07-15 10:36:28,003 - owrx.fft - DEBUG - Spectrum thread initialized successfully.
Jul 15 10:36:28 rds-2 openwebrx[3262]: 2020-07-15 10:36:28,004 - csdr.csdr - DEBUG - Command = nc -v 127.0.0.1 1001 | csdr fft_cc 16384 11377.777777777777 | csdr logaveragepower_cf -70 16384 10 | csdr fft_exchange_sides_ff 16384 | csdr compress_fft_adpcm_f_u8 16384
Jul 15 10:36:28 rds-2 openwebrx[3262]: client connection establised
Jul 15 10:36:28 rds-2 openwebrx[3262]: Connection to 127.0.0.1 1001 port [tcp/customs] succeeded!
Jul 15 10:36:28 rds-2 openwebrx[3262]: 2020-07-15 10:36:28,920 - csdr.csdr - DEBUG - Command = nc -v 127.0.0.1 1001 | csdr shift_addfast_cc --fifo /websdr/openwebrx_pipe_139939302563232_shift_pipe | csdr fir_decimate_cc 85 0.0017647058823529412 HAMMING | csdr bandpass_fir_fft_cc --fifo /websdr/openwebrx_pipe_139939302563232_bpf_pipe 0.0265625 HAMMING | csdr squelch_and_smeter_cc --fifo /websdr/openwebrx_pipe_139939302563232_squelch_pipe --outfifo /websdr/openwebrx_pipe_139939302563232_smeter_pipe 5 20 | csdr realpart_cf | csdr fractional_decimator_ff 1.003921568627451 | csdr agc_ff | csdr limit_ff | csdr convert_f_s16 | csdr encode_ima_adpcm_i16_u8
Jul 15 10:36:28 rds-2 openwebrx[3262]: 2020-07-15 10:36:28,920 - owrx.websocket - ERROR - Exception in websocket handler handleTextMessage()
Jul 15 10:36:28 rds-2 openwebrx[3262]: Traceback (most recent call last):
Jul 15 10:36:28 rds-2 openwebrx[3262]:   File "/usr/lib/python3/dist-packages/owrx/websocket.py", line 203, in read_loop
Jul 15 10:36:28 rds-2 openwebrx[3262]:     self.messageHandler.handleTextMessage(self, message)
Jul 15 10:36:28 rds-2 openwebrx[3262]:   File "/usr/lib/python3/dist-packages/owrx/connection.py", line 144, in handleTextMessage
Jul 15 10:36:28 rds-2 openwebrx[3262]:     self.startDsp()
Jul 15 10:36:28 rds-2 openwebrx[3262]:   File "/usr/lib/python3/dist-packages/owrx/connection.py", line 244, in startDsp
Jul 15 10:36:28 rds-2 openwebrx[3262]:     self.getDsp().start()
Jul 15 10:36:28 rds-2 openwebrx[3262]:   File "/usr/lib/python3/dist-packages/owrx/dsp.py", line 141, in start
Jul 15 10:36:28 rds-2 openwebrx[3262]:     self.dsp.start()
Jul 15 10:36:28 rds-2 openwebrx[3262]:   File "/usr/lib/python3/dist-packages/csdr/csdr.py", line 787, in start
Jul 15 10:36:28 rds-2 openwebrx[3262]:     self.process = subprocess.Popen(command, stdout=out, shell=True, start_new_session=True, env=my_env)
Jul 15 10:36:28 rds-2 openwebrx[3262]:   File "/usr/lib/python3.7/subprocess.py", line 775, in __init__
Jul 15 10:36:28 rds-2 openwebrx[3262]:     restore_signals, start_new_session)
Jul 15 10:36:28 rds-2 openwebrx[3262]:   File "/usr/lib/python3.7/subprocess.py", line 1412, in _execute_child
Jul 15 10:36:28 rds-2 openwebrx[3262]:     errpipe_read, errpipe_write = os.pipe()
Jul 15 10:36:28 rds-2 openwebrx[3262]: OSError: [Errno 24] Too many open files
Jul 15 10:36:38 rds-2 openwebrx[3262]: 2020-07-15 10:36:38,521 - csdr.csdr - WARNING - queue /websdr/openwebrx_pipe_139939302563232_bpf_pipe never successfully opened - thread leak!
Jul 15 10:36:38 rds-2 openwebrx[3262]: 2020-07-15 10:36:38,521 - csdr.csdr - WARNING - queue /websdr/openwebrx_pipe_139939302563232_shift_pipe never successfully opened - thread leak!
Jul 15 10:36:38 rds-2 openwebrx[3262]: 2020-07-15 10:36:38,521 - csdr.csdr - WARNING - queue /websdr/openwebrx_pipe_139939302563232_squelch_pipe never successfully opened - thread leak!
Jul 15 10:36:38 rds-2 openwebrx[3262]: closing client socket
Jul 15 10:36:38 rds-2 openwebrx[3262]: 2020-07-15 10:36:38,522 - owrx.websocket - ERROR - Exception in websocket handler handleTextMessage()
Jul 15 10:36:38 rds-2 openwebrx[3262]: Traceback (most recent call last):
Jul 15 10:36:38 rds-2 openwebrx[3262]:   File "/usr/lib/python3/dist-packages/owrx/websocket.py", line 203, in read_loop
Jul 15 10:36:38 rds-2 openwebrx[3262]:     self.messageHandler.handleTextMessage(self, message)
Jul 15 10:36:38 rds-2 openwebrx[3262]:   File "/usr/lib/python3/dist-packages/owrx/connection.py", line 163, in handleTextMessage
Jul 15 10:36:38 rds-2 openwebrx[3262]:     self.setSdr(profile[0])
Jul 15 10:36:38 rds-2 openwebrx[3262]:   File "/usr/lib/python3/dist-packages/owrx/connection.py", line 202, in setSdr
Jul 15 10:36:38 rds-2 openwebrx[3262]:     self.getDsp()
Jul 15 10:36:38 rds-2 openwebrx[3262]:   File "/usr/lib/python3/dist-packages/owrx/connection.py", line 283, in getDsp
Jul 15 10:36:38 rds-2 openwebrx[3262]:     self.dsp = DspManager(self, self.sdr)
Jul 15 10:36:38 rds-2 openwebrx[3262]:   File "/usr/lib/python3/dist-packages/owrx/dsp.py", line 79, in __init__
Jul 15 10:36:38 rds-2 openwebrx[3262]:     mode = Modes.findByModulation(self.props["start_mod"])
Jul 15 10:36:38 rds-2 openwebrx[3262]:   File "/usr/lib/python3/dist-packages/owrx/modes.py", line 88, in findByModulation
Jul 15 10:36:38 rds-2 openwebrx[3262]:     modes = [m for m in Modes.getAvailableModes() if m.modulation == modulation]
Jul 15 10:36:38 rds-2 openwebrx[3262]:   File "/usr/lib/python3/dist-packages/owrx/modes.py", line 80, in getAvailableModes
[... cut ...]
Jul 15 10:37:36 rds-2 openwebrx[3262]:     process = subprocess.Popen([command, "--version"], stdout=subprocess.PIPE)
Jul 15 10:37:36 rds-2 openwebrx[3262]:   File "/usr/lib/python3.7/subprocess.py", line 728, in __init__
Jul 15 10:37:36 rds-2 openwebrx[3262]:     errread, errwrite) = self._get_handles(stdin, stdout, stderr)
Jul 15 10:37:36 rds-2 openwebrx[3262]:   File "/usr/lib/python3.7/subprocess.py", line 1352, in _get_handles
Jul 15 10:37:36 rds-2 openwebrx[3262]:     c2pread, c2pwrite = os.pipe()
Jul 15 10:37:36 rds-2 openwebrx[3262]: OSError: [Errno 24] Too many open files
`

jketterl, 2020-07-20:
Alright, I traced back some of the file descriptors that were accumulating over time, and I identified two pretty similar leaks in the usage of the python "multiprocessing" library. They are both fixed in my usage now, but of course this will need some longer testing.

T-Shilov, 2020-07-20:
We hope this is the final solution to this riddle.
Thank you Jakob for your attentiveness to our requests and good work!

jketterl, 2020-07-20:
final solution? riddle?

it's progress in software development, nothing more.

T-Shilov, 2020-07-21:
> final solution?

I do not know this. I temporary was saved from this bug through by a daily reboot of the system.

jketterl, 2020-08-07:
I found and fixed some more file descriptor leaks in the background decoding services and secondary demodulators, from what i can see it is pretty stable now. I'm currently hoping I didn't cause to many issues in other places, I guess those will show up soon enough.

Unfortunately, the station I'm maintaining that ran into this problem seems to be experiencing hardware issues, I cannot currently test this in the long run. Any feedback here is appreciated.

T-Shilov, 2020-08-07:
Dear Jakob, 
Thank you very much for being attentive to the requests of your users. 
Your project is getting better and better. At the moment, it is the best of those that I know, it has rich functionality, great prospects, and it is a pleasure to use it :)
I also hope that the various shortcomings that usually arise in an actively developing project will be eliminated over time.

Best regards,
   Shilov

jketterl, 2020-08-26:
From my personal monitoring at multiple stations, I don't see any file descriptor or thread leaks any more. I'm going to close this for now, all fixes will be in the next release.

closed 2020-08-26 by jketterl