jketterl/openwebrx #148: Compiler error on Pi4 and Buster

loewal, 2020-07-08:
Can somebody help please?
/tmp/cclfSInx.s:2634: Error: selected processor does not support `veor q4,q4' in ARM mode
/tmp/cclfSInx.s:2635: Error: selected processor does not support `veor q5,q5' in ARM mode
/tmp/cclfSInx.s:2636: Error: selected processor does not support `vld2.32 {q0-q1},[lr]!' in ARM mode
/tmp/cclfSInx.s:2637: Error: selected processor does not support `vld1.32 {q2},[r4]!' in ARM mode
/tmp/cclfSInx.s:2638: Error: selected FPU does not support instruction -- `vmla.f32 q4,q0,q2'
/tmp/cclfSInx.s:2639: Error: selected FPU does not support instruction -- `vmla.f32 q5,q1,q2'
/tmp/cclfSInx.s:2642: Error: selected processor does not support `vst1.32 {q4},[r7]' in ARM mode
/tmp/cclfSInx.s:2643: Error: selected processor does not support `vst1.32 {q5},[r6]' in ARM mode
make: *** [Makefile:607: libcsdr_la-libcsdr.lo] Fout 1

jketterl, 2020-07-08:
What version of csdr are you using? Is this the 64bit Raspberry Pi OS?

loewal, 2020-07-08:
Yes, that's correct.

loewal, 2020-07-08:
On Pi4. I could install it the easy way (apt-get..)

jketterl, 2020-07-08:
Please check if you're using my fork of csdr: https://github.com/jketterl/csdr

The version by ha7ilm unfortunately does not cover 64bit ARM, so that might explain the errors you're getting.

loewal, 2020-07-08:
I checked and I used the instructions in the manual setup. 
Tried again with instructions on https://github.com/jketterl/csdr
The same helas!

jketterl, 2020-07-08:
That's odd. The `veor` instructions are defined for `__arm__` only, so they should not be included by the preprocessor an aarch64. I will need to get a copy of Raspberry Pi 64bit and test it myself to see what's going wroing.

Please keep in mind though that the 64bit version of Raspberry Pi OS is still experimental, so this might as well be some issue there.

loewal, 2020-07-08:
And yes! I switched to v7l+ and all went well.

jketterl, 2020-07-08:
Yes, that works well. 64bit *should* work too, but I have only been testing on Ubuntu 64 bit so far.

loewal, 2020-07-08:
Thanks by the way!

jketterl, 2020-07-09:
I just downloaded the Raspberry Pi OS 64-bit beta image from [here](https://www.raspberrypi.org/forums/viewtopic.php?t=275370), and I cannot reproduce. The compile step went through as expected, only pulling in the armv8 64bit neon code. Resulting binary seems to be working (I just did a `csdr benchmark`).

Did you install any 32bit compatibility packages or something similar?

jketterl, 2020-07-09:
They might as well finally adress https://github.com/pothosware/SoapySDRPlay/issues/60 which would probably make additional controls obsolete.

loewal, 2020-07-09:
I downloaded it from here: https://www.raspberrypi.org/downloads/raspberry-pi-os/
But as you (and me too now 😖 ) can see it's 32-bit. The only thing I did was adding one line in /boot/config.txt: **arm_64bit=1**
And then **kernel8.img** is used.
That works.
uname -a output:
**Linux Pi4 5.4.44-v8+ #1320 SMP PREEMPT Wed Jun 3 16:20:05 BST 2020 aarch64 GNU/Linux**
This is not the right way?

loewal, 2020-07-09:
About the modules...I don't know anymore. I followed your step-by-step instructions.

loewal, 2020-07-09:
Do you think it's worth wile to use the 64-bit environment?

jketterl, 2020-07-10:
I see, I'm not too sure about that method. Basciall that means that you're running a 64bit kernel, but anything else is still compiled for 32bit, not sure how that works out for the applications. That includes the compiler and the system include files, and I guess that's the issue here. Not sure if that could be fixed, but I guess that there's probably other issues there as well. There's probably reasons why there's a completely separate 64bit image available.

As for OpenWebRX: I don't see that it profits from the 64bit environment. The performance is pretty much the same, and the memory consumption is not near anything that would require a 64bit address space. I have added the ARMv8 support to csdr since it's clearly going to be the way forward, but there's no real issues with the 32bit environment, at least none that I'm aware of.

loewal, 2020-07-10:
> I see, I'm not too sure about that method. Bascially that means that you're running a 64bit kernel, but anything else is still compiled for 32bit, not sure how that works out for the applications. That includes the compiler and the system include files, and I guess that's the issue here. Not sure if that could be fixed, but I guess that there's probably other issues there as well. There's probably reasons why there's a completely separate 64bit image available.

Ok, thanks Jakob.
I will revert to 32-bits.

Grtz. Loe

Loe Walter
Insula 103
6416 BZ  Heerlen
☎️ +31-6-24476963

closed 2020-07-10 by jketterl