jketterl/openwebrx #149: WSJT-X 2.2.2

nevrrmind, 2020-07-08:
V2.2.2 is online :)
https://physics.princeton.edu/pulsar/K1JT/wsjtx-2.2.2.tgz

jketterl, 2020-07-08:
That's good! Does it work with OpenWebRX?

nevrrmind, 2020-07-08:
Still waiting for compiling to finish...

loewal, 2020-07-08:
Runs perfect on my Pi4!

nevrrmind, 2020-07-08:
a2x: ERROR: "xsltproc" -param man.endnotes.list.enabled 0 -param man.endnotes.are.numbered 0 --stringparam callout.graphics 0 --stringparam navig.graphics 0 --stringparam admon.textlabel 1 --stringparam admon.graphics 0  "/etc/asciidoc/docbook-xsl/manpage.xsl" "/opt/wsjtx-2.2.2/build/wsjtx-prefix/src/wsjtx-build/manpages/man/man1/wsjtx.1.xml" returned non-zero exit status 127

> apt install xsltproc (Maybe you could add it to the install instructions.)

nevrrmind, 2020-07-08:
> Modulation "JT65" not supported. Please check requirements
Modulation "JS8Call" not supported. Please check requirements
Modulation "FT8" not supported. Please check requirements
Modulation "FT4" not supported. Please check requirements
Modulation "WSPR" not supported. Please check requirements
> Modulation "JT65" not supported. Please check requirements
Modulation "JS8Call" not supported. Please check requirements
Modulation "FT8" not supported. Please check requirements
Modulation "FT4" not supported. Please check requirements
Modulation "WSPR" not supported. Please check requirements

Maybe i missed something...

jketterl, 2020-07-08:
Please check the availability of both the `jt9` and `wsprd` commands. They are the requirements for wsjt-x decoding.

nevrrmind, 2020-07-08:
Very simple solution: I forgot "make install" 🙄

jketterl, 2020-07-08:
ah. well at least it's easy to fix :grinning:

jketterl, 2020-07-08:
`xsltproc` added to manual installation procedure :)

nevrrmind, 2020-07-08:
So I can summarize= V2.2.2 is running. :)
>Device model : Odroid C4 (aarch64)
>PRETTY_NAME="Debian GNU/Linux 10 (buster)"
>Linux 4.9.218+ aarch64

closed 2020-07-08 by nevrrmind