jketterl/openwebrx #150: Not all dongles are started

T-Shilov, 2020-07-08:
I use 6 pieces of RTL-SDR v3 dongles. They are brand new and serviceable.

But because of this problem https://github.com/jketterl/openwebrx/issues/146 I have to restart OpenWebRX.
And then a second problem was discovered: not all dongles are started. Some random dongle does not start.
See the app for an example.

[image]
[image]

And not the same dongles, but each time a different one.
The same problem was detected after rebooting the system.
So I need again and again to reload OpenWebRX to start all 6 dongles.

This is a serious problem. I ask you to eliminate this.

OpenWebRX 0.19.1 and Debian 10/64.
CPU i3. RAM 8 GB.

jketterl, 2020-07-09:
Please check the logfile for messages on what happened when the devices were started. I am suspecting messages along the lines of `no device found` or `device could not be opened`.

jketterl, 2020-07-18:
Anything? I'm not going to keep this open unless there's some kind of information available as to what part of OpenWebRX would cause this.

T-Shilov, 2020-07-18:
Ok, close this for now. If it appears again, I'll let you know.

jketterl, 2020-07-25:
I'm asking because there is a very high chance that these problems are not actually caused by OpenWebRX. I have seen many dongles fail due to power issues, either resulting in the dongle being ejected by the kernel, or by responding with errors upon opening. In such cases, OpenWebRX is merely responding to those errors by disabling the device, but there's nothing it could do to revive them, they usually need a power cycle.

closed 2020-08-26 by jketterl

T-Shilov, 2020-08-27:
Jakob, excuse me, I forgot to tell you how I temporarily solved this problem.
My dongles are of high quality and their supply voltage is excellent.
But with a common restart of OpenWebRX, some dongls do not start:

`systemctl restart openwebrx`

And every time they are different. So the problem is not in them.
Then I paused before running the OpenWebRX:

```
systemctl stop openwebrx
sleep 10
systemctl start openwebrx
```

And then all the dongles of everything start reliably.

5 seconds is sometimes not enough, you need at least 10 seconds.

I hope this workaround will  hint you with the best solution to this problem.

Best regards

Shilov