jketterl/openwebrx #151: Setting if_gain on SDRPlay devices

Challado, 2020-07-09:
SDRPlay have a parameter called "if_gain" that, as the name says, define the gain 
SoapySDRUtil --probe
######################################################
## Soapy SDR -- the SDR abstraction library
######################################################

Probe device

RtApiAlsa::getDeviceInfo: snd_pcm_open error for device (default), No such file or directory.

[INFO] [UHD] linux; GNU C++ version 8.2.0; Boost_106700; UHD_3.13.1.0-3
[INFO] devIdx: 0
[INFO] hwVer: 1

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=SDRplay
  hardware=B0001P0002
  sdrplay_api_api_version=0.000000
  sdrplay_api_hw_version=1

----------------------------------------------------
-- Peripheral summary
----------------------------------------------------
  Channels: 1 Rx, 0 Tx
  Timestamps: NO
  Other Settings:
     * RF Gain Select - RF Gain Select
       [key=rfgain_sel, default=1, type=string, options=(0, 1, 2, 3)]
     * IQ Correction - IQ Correction Control
       [key=iqcorr_ctrl, default=true, type=bool]
     * AGC Setpoint - AGC Setpoint (dBfs)
       [key=agc_setpoint, default=-30, type=int, range=[-60, 0]]

----------------------------------------------------
-- RX Channel 0
----------------------------------------------------
  Full-duplex: YES
  Supports AGC: YES
  Stream formats: CS16, CF32
  Native format: CS16 [full-scale=32767]
  Antennas: RX
  Corrections: DC removal
  Full gain range: [0, 42] dB
    IFGR gain range: [20, 59] dB
    RFGR gain range: [0, 3] dB
  Full freq range: [0.01, 2000] MHz
    RF freq range: [0.01, 2000] MHz
    CORR freq range:  MHz
  Sample rates: 0.0625, 0.125, 0.25, 0.5, 1, 6, 7, 8, 9, 10 MSps
  Filter bandwidths: 0.2, 0.3, 0.6, 1.536, 5, 6, 7, 8 MHz

ISoapySDRUtil --probe
######################################################
## Soapy SDR -- the SDR abstraction library
######################################################

Probe device

RtApiAlsa::getDeviceInfo: snd_pcm_open error for device (default), No such file or directory.

[INFO] [UHD] linux; GNU C++ version 8.2.0; Boost_106700; UHD_3.13.1.0-3
[INFO] devIdx: 0
[INFO] hwVer: 1

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=SDRplay
  hardware=B0001P0002
  sdrplay_api_api_version=0.000000
  sdrplay_api_hw_version=1

----------------------------------------------------
-- Peripheral summary
----------------------------------------------------
  Channels: 1 Rx, 0 Tx
  Timestamps: NO
  Other Settings:
     * RF Gain Select - RF Gain Select
       [key=rfgain_sel, default=1, type=string, options=(0, 1, 2, 3)]
     * IQ Correction - IQ Correction Control
       [key=iqcorr_ctrl, default=true, type=bool]
     * AGC Setpoint - AGC Setpoint (dBfs)
       [key=agc_setpoint, default=-30, type=int, range=[-60, 0]]

----------------------------------------------------
-- RX Channel 0
----------------------------------------------------
  Full-duplex: YES
  Supports AGC: YES
  Stream formats: CS16, CF32
  Native format: CS16 [full-scale=32767]
  Antennas: RX
  Corrections: DC removal
  Full gain range: [0, 42] dB
    IFGR gain range: [20, 59] dB
    RFGR gain range: [0, 3] dB
  Full freq range: [0.01, 2000] MHz
    RF freq range: [0.01, 2000] MHz
    CORR freq range:  MHz
  Sample rates: 0.0625, 0.125, 0.25, 0.5, 1, 6, 7, 8, 9, 10 MSps
  Filter bandwidths: 0.2, 0.3, 0.6, 1.536, 5, 6, 7, 8 MHz

If possible to put these parameter on openwebrx, I appreciate so much

jketterl, 2020-07-09:
The two gain stages of the SDRPlay devices are available, but they are not straightforward to use. As described in the documentation, you can set separate gains by passing a SoapySDR string to the `rf_gain` parameter, e.g. `"rf_gain"="RFGR=3,IFGR=20"`.

https://github.com/jketterl/openwebrx/wiki/SDR-device-and-profile-configuration#rf_gain

Challado, 2020-07-09:
As I see on SDRPlay doc (correct me if i'm wrong), the if_gain is another
parameter that maximizes the overall input gain of rf device. The RF_gain
is fine tuning over the intermediate (IFGR) frequency and radio frequency
(RFGR), as I understand on these topic.
https://www.sdrplay.com/community/viewtopic.php?t=4131
Please correct me if I'm wrong...

jketterl, 2020-07-09:
Well, unfortunately there is no `if_gain` parameter exposed on the SoapySDR module, as confirmed by the probe output you posted above, so there's nothing I could do to make it controllable.

To my understanding, IFGR means "IF gain reduction", so as far as I'm concerned, that seems to be what you're asking for, even though for reasons beyond my explanation, it will behave reverse to what you'd expect from a gain value ("reduction").

Challado, 2020-07-09:
Oh, *sorry for my mistake*
The parameter that I need to citate is rfgain_sel.
Sorry for my stupid dumb error.

jketterl, 2020-07-09:
Sorry, but that's a no. SDRPlay need to sort out their gain settings beforehand. From what I understand, the `rfgain_sel` parameter is synthetic and implements a set of presets for the other gain settings. That functionality could easily implemented using the regular gain API, instead of messing things up further with a custom setting.

jketterl, 2020-07-09:
They might as well finally adress https://github.com/pothosware/SoapySDRPlay/issues/60 which would probably make additional controls obsolete.

closed 2020-07-18 by jketterl