jketterl/openwebrx #152: CSDR on Orange Pi Pc2

loewal, 2020-07-10:
Following error....compiling your csdr:
o libcsdr_la-fft_rpi.lo libcsdr_la-libcsdr.lo libcsdr_la-libcsdr_gpl.lo libcsdr_la-fastddc.lo libcsdr_la-ima_adpcm.lo -L/usr/local/lib -lfftw3f -lm
libtool: link: gcc -shared  -fPIC -DPIC  .libs/libcsdr_la-fft_fftw.o .libs/libcsdr_la-fft_rpi.o .libs/libcsdr_la-libcsdr.o .libs/libcsdr_la-libcsdr_gpl.o .libs/libcsdr_la-fastddc.o .libs/libcsdr_la-ima_adpcm.o   -L/usr/local/lib /usr/local/lib/libfftw3f.a -lm  -g -O3 -march=armv8-a -mtune=cortex-a72   -Wl,-soname -Wl,libcsdr-0.16.1.so -o .libs/libcsdr-0.16.1.so
/usr/bin/ld: /usr/local/lib/libfftw3f.a(tensor7.o): relocation R_AARCH64_ADR_PREL_PG_HI21 against symbol `fftwf_dimcmp' which may bind externally can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: /usr/local/lib/libfftw3f.a(tensor7.o): in function `fftwf_tensor_compress_contiguous.part.1':
tensor7.c:(.text+0x48c): dangerous relocation: unsupported relocation
/usr/bin/ld: /usr/local/lib/libfftw3f.a(tensor7.o): relocation R_AARCH64_ADR_PREL_PG_HI21 against symbol `fftwf_dimcmp' which may bind externally can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: /usr/local/lib/libfftw3f.a(tensor7.o): in function `fftwf_tensor_compress':
tensor7.c:(.text+0x678): dangerous relocation: unsupported relocation
collect2: error: ld returned 1 exit status
make: *** [Makefile:538: libcsdr.la] Fout 1

I could compile a different csdr, but needed to alter the PARAMS_NEON, but the result was, that sound came after 6 seconds, waterfall immediately. Changing parameters to minimal values didn't change things.

Could you take a look at the error Jakob?

loewal, 2020-07-10:
Addition:
Linux orange 4.19.63-sunxi64 #5.92 SMP Fri Aug 2 00:18:27 CEST 2019 aarch64 GNU/Linux

jketterl, 2020-07-11:
Well, I have had a look, and honestly, I don't have no idea what that means. Could you fill me in on what distribution and version thereof is running? I'll try to reproduce if possible...

loewal, 2020-07-12:
The 'different' CSDR is the one I copied from: https://github.com/ha7ilm/csdr.git
I had to change the makefile like this:
PARAMS_SSE = $(call cpufeature,sse,-msse) $(call cpufeature,sse2,-msse2) $(call cpufeature,sse3,-msse3) $(call cpufeature,sse4a,-msse4a) $(call cpufeature,sse4_1,-msse4.1) $(call cpufeature,sse4_2,-msse4.2 -msse4) -mfpmath=sse
**_**PARAMS_NEON = -O3 -mcpu=cortex-a53 -march=armv8-a+crc+crypto -fomit-frame-pointer -funroll-loops -Wformat=0**_**
#tnx Jan Szumiec for the Raspberry Pi support
PARAMS_RASPI = -mfloat-abi=hard -mcpu=arm1176jzf-s -mfpu=vfp -funsafe-math-optimizations -Wformat=0
PARAMS_ARM = $(if $(call cpufeature,BCM2708,dummy-text),$(PARAMS_RASPI),$(PARAMS_NEON))
PARAMS_SIMD = $(if $(call cpufeature,sse,dummy-text),$(PARAMS_SSE),$(PARAMS_ARM))

So the PARAMS_NEON are different from his original. I found this on the internet. It compiles well, but sound comes only after 15 to 20 seconds......

loewal, 2020-07-12:
Then I tried yours again...and wanted to change that same PARAMS_NEON value, but didn't find it. Compiling it as it is gave again this error:
/usr/bin/ld: /usr/local/lib/libfftw3f.a(tensor7.o): relocation R_AARCH64_ADR_PREL_PG_HI21 against symbol `fftwf_dimcmp' which may bind externally can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: /usr/local/lib/libfftw3f.a(tensor7.o): in function `fftwf_tensor_compress_contiguous.part.1':
tensor7.c:(.text+0x48c): dangerous relocation: unsupported relocation
/usr/bin/ld: /usr/local/lib/libfftw3f.a(tensor7.o): relocation R_AARCH64_ADR_PREL_PG_HI21 against symbol `fftwf_dimcmp' which may bind externally can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: /usr/local/lib/libfftw3f.a(tensor7.o): in function `fftwf_tensor_compress':
tensor7.c:(.text+0x678): dangerous relocation: unsupported relocation
collect2: error: ld returned 1 exit status
make: *** [Makefile:538: libcsdr.la] Fout 1

jketterl, 2020-07-12:
No I mean, what distribution are you running? Ubuntu, Debian, something else?

loewal, 2020-07-12:
Debian Buster

jketterl, 2020-07-12:
I just noticed the paths for your fftw3 point to `/usr/local` - did you compile that from source? If so, did that build produce any shared libraries? I suppose something is off there since the build attempts to link against `libfftw3f.a`, which is a static library and won't work. It should link against `libfftw3.so`, which is represented on the command line as `-lfftw3f`.

jketterl, 2020-07-12:
Yes, that seems to be exactly the issue. I just tried to compile against a version of FFTW3 that was compiled from source and ended up with the same error. Please recompile your FFTW3 with `--enable-shared`, that should fix it.

I have updated the readme to reflect this. Thanks for bringing this up, I never tried that myself :slightly_smiling_face:

loewal, 2020-07-13:
I compiled and installed FFTW 3.3.4 as follows:
**./configure --enable-float --enable-sse --enable-shared**
Then downloaded CSDR as described in your manual installation.
Exactly the same error helas.
Then downloaded and installed the other CSDR (https://github.com/ha7ilm/csdr.git)
Everything compiles fine, but then there is again that timelag of about 15 seconds for the first sound when the screen is already running.
Then: apt install libfftw3 --reinstall and compiled. Same error!
I think, with my little knowledge helas, that I am doing something wrong.......

jketterl, 2020-07-13:
Well I don't have any real advice, but... a few notes:

- `--enable-sse` should not work on aarch64. I received an error for that, but of course if you configured previously with another set of parameters, and overlook that error, you'd end up without anything new.
- If you're trying to get this working with version 3.3.4 because of https://github.com/ha7ilm/openwebrx/issues/139: There's not point in downgrading FFTW3, the change was in the way the debian packages are built, not in the libraries itself.
- Once you have a manually compiled version on your system, that will take precedence over any packages, so installing the `libfftw3` package after the fact won't have any effect. You'd have to remove the manual installation first.
- Installing `libfftw3` alone is not sufficient, you will need `libfftw3-dev` for the header and package information.

Other than that, I just tried it with version 3.3.4 and it worked perfectly fine. You would need to `make clean` and re-run the `configure` script in csdr after the FFTW3 issues are resolve, but since you stated that you downloaded csdr (I'm assuming a fresh copy), that shouldn't be necessary.

loewal, 2020-07-13:
Thnx Jakob,
I have a Pi4 running fine and an AMD64 as well, so I don't need it actually, but the fact alone that errors occur, keeps me searching for a solution.
Maybe I do an image-restore and give it a try again.

For now: thanks for the beautiful piece of software!

jketterl, 2020-07-13:
I see, well at least the sse isn't the problem then... I don't really know what's wrong at this point. I've tried various combinations of installed packages and compiled FFTW3, and as long as I keep `--enable-shared --enable-float` on the fftw3 configure call, it works out fine.

There's two things you can check:
- Is there any shared libraries? `ls /usr/local/lib/libfftw3f.so.*`
- What's the contents of the fftw3 pkg-config file `/usr/local/lib/pkgconfig/fftw3f.pc`? Here's mine for reference:

```
prefix=/usr/local
exec_prefix=${prefix}
libdir=${exec_prefix}/lib
includedir=${prefix}/include

Name: FFTW
Description: fast Fourier transform library
Version: 3.3.4
Libs: -L${libdir} -lfftw3f 
Libs.private: -lm
Cflags: -I${includedir}
```

loewal, 2020-07-13:
All correct, except Version: 3.3.8
Also I saw, that there are two versions....3.3.7 and 3.3.8 in /usr/local/lib/
ll /usr/local/lib/*fftw*
-rw-r--r-- 1 root 1777086 jul 13 08:45 /usr/local/lib/libfftw3.a
-rw-r--r-- 1 root 1805378 sep  7  2019 /usr/local/lib/libfftw3f.a
-rwxr-xr-x 1 root     889 sep  7  2019 /usr/local/lib/libfftw3f.la
-rwxr-xr-x 1 root     942 jul 13 08:45 /usr/local/lib/libfftw3.la
lrwxrwxrwx 1 root      17 jul 13 08:45 /usr/local/lib/libfftw3.so -> libfftw3.so.3.5.8
lrwxrwxrwx 1 root      17 jul 13 08:45 /usr/local/lib/libfftw3.so.3 -> libfftw3.so.3.5.8
lrwxrwxrwx 1 root      13 jul 13 07:28 /usr/local/lib/libfftw3.so.3.5.7 -> libfftw3.so.3
-rwxr-xr-x 1 root  830672 jul 13 08:45 /usr/local/lib/libfftw3.so.3.5.8
root@orange:~#
[image]

jketterl, 2020-07-13:
Your listing is clearly missing libfftw3f.so - notice the trailing "f", that's the version for floating point arithmetic, which is the one that csdr uses. There is a libfftw3f.a, which is a static lib, but not the shared version.

loewal, 2020-07-14:
I let Midnight Commmander search:
[image]

How can I get a shared version?.......

loewal, 2020-07-14:
I download http://www.fftw.org/fftw-3.3.8.tar.gz
compiled with
make clean &&

./configure --prefix=/usr    \
            --enable-shared  \
            --enable-threads \
            --enable-long-double &&
make
and
make install
Then did the manual installation as described by you.
Still getting the same error....
Do I have to specify the librarypath explicitly?

loewal, 2020-07-14:
The search like before gives the same result

loewal, 2020-07-14:
Even stranger now:
I removed libfftw3-dev with apt remove. Now the other CSDR gives the same error.
Installed libfftw3 again and compiled ok!
Then did your version...still that old error....

jketterl, 2020-07-14:
As I said before, you need to compile FFTW3 with `./configure --enable-shared --enable-float`.

Setting the prefix to `/usr` is a bad idea btw, in that location it will collide with your distributions' package file... That's probably a pretty big mess by now. Also, installations in `/usr/local` will take precedence over `/usr`, so as long as you still have the files present in `/usr/local`, installing something in `/usr` (no matter if it's packages or prefixed) won't have much of an effect.

loewal, 2020-07-14:
Solved!!!
I deleted libfftw3* from /usr/local/lib and compiled csdr again with the right parameters.
Didn't work....until I did a ldconfig.
Thanks Jakob

jketterl, 2020-07-18:
Ah. Yeah, `ldconfig` seems to be a necessary step on many distributions when installing in `/usr/local/`. There's a big discussion on some linux mailing list that I don't really comprehend about why there's a difference.

Anyway, glad you got it sorted. Enjoy! :)

closed 2020-07-18 by jketterl

loewal, 2020-07-18:
Thanks very much!