jketterl/openwebrx #153: repeaterbook integration

cTurtle98, 2020-07-15:
I noticed that repeaterbook has an api where you can search by frequency
https://www.repeaterbook.com/wiki/doku.php?id=api

it would be cool if you could click on a frequency in the waterfall and have it search repeaterbook and see if that is a known frequency or not.

it would be kinda difficult to implement because centering the tuner on the carrier frequency is a little difficult but it would be cool

jketterl, 2020-07-18:
If they would finally merge their database and provide wordwide coverage, I may be able to do it. Unfortunately, I live outside the US, so that API is completely useless, which massively blocks the development process. Also, my motivation to include features that only work in the US is very limited.

mgrennan, 2022-08-31:
I’m surprised you don’t know it does and supplies many outputs.
https://www.repeaterbook.com/row_repeaters/Display_SS.php?state_id=DE&band=14&loc=%25&call=%25&use=%25

scannerbrightly, 2023-11-12:
Jakob, 

Is there a technical limitation with the ReapeterBook database, or is it something else? I see all of Europe in there, [Italy 2m for example](https://www.repeaterbook.com/row_repeaters/Display_SS.php?state_id=IT&band=14&loc=%&call=%&use=%)

You said 'merge database'. Is it the 'US' and 'Rest of World' databases that are the problem? It appears from the [API page](https://www.repeaterbook.com/wiki/doku.php?id=api) that the only difference is the URL, but the formatting is the same. I would love to integrate this somehow.

jketterl, 2023-11-12:
> I’m surprised you don’t know it does and supplies many outputs.

You may find that there's a time period of two years between my last comment and yours. I'm pretty sure these things were not available back in 2020. Even if they were available, you can't expect me to know everything. That's why we have an issue tracker, if you know about these things, you can just drop them off here, and then we can work from there.

> You said 'merge database'. Is it the 'US' and 'Rest of World' databases that are the problem? It appears from the [API page](https://www.repeaterbook.com/wiki/doku.php?id=api) that the only difference is the URL, but the formatting is the same. I would love to integrate this somehow.

Yes, the split is a problem. I'm not sure why they're keeping that, but for this project it means that I would need to provide the ability for operators to switch between those two, which would mean extra effort.