jketterl/openwebrx #154: Two questions about OpenWebRX

T-Shilov, 2020-07-15:
Dear Jakob!
This is not a complaint, but only ordinary questions.

Please, say Me:

1. How do I use this piece of configuration **config_webrx.py**to add a photo or picture?
```
photo_desc="""
You can add your own background photo and receiver information.<br />
```
I do not understand this  mysterious syntax :)
Please show a practical example of how to use it.

2. Does API exist in OpenWebRX? I have long wanted to connect a chat to it.

jketterl, 2020-07-16:
First of all, there's a Mailing List available for general questions and support. Please use that in the future.

There is currently no way to configure images, and I'm not officially supporting this. While some people may say this was a feature of the old version of OpenWebRX (before I started working on it), I say it can hardly be considered a feature if you have to replace files in your local workspace copay. Nevertheless, that's how it worked before.

The problems start when you get into properly bundling and distributing software: Now people cannot easily replace images any more. And if they do, the package manager will undo their changes on every update. As such, changing any of the images is currently not officially supported.

The text you quoted still comes from the original, 6 years ago.

As for the API: What kind of API would you wish for? You'd need to be way more specific than that. There's no general API that you'd throw onto anything to make it compatible with anything you wish for.

As for adding a chat to OpenWebRX: There's currently no support for that.

jketterl, 2020-07-16:
Here's the documentation of the original: https://github.com/ha7ilm/openwebrx/wiki/Customizing-OpenWebRX-for-your-station

If you know what to change, you can still do it, but as I said, it's not official, so I'm not going to spill the beans here.

T-Shilov, 2020-07-16:
I understood you, Jakob.

About API. I would like to get an API that allows third party chat and forum to be connected.
For example, CMS Drupal, which has [chat](https://www.drupal.org/project/drupalchat) and forum plugins.

jketterl, 2020-07-18:
Oh. That's not an API, that's just custom plugins for drupal. That's a completely closed environment and you won't see any of these plugins outside of drupal.

T-Shilov, 2020-07-18:
Well, this is understandable. But can Drupal itself be connected to OpenWebRX ?

jketterl, 2020-07-25:
Well, from my point of view those systems are following a completely different purpose, so at least I cannot really imagine a connection as of now, especially on such a general level.

closed 2020-08-05 by jketterl