jketterl/openwebrx #156: A waterfall suddenly stopped on one RTL-SDR

T-Shilov, 2020-07-25:
Hello, Jakob,

Today a waterfall suddenly stopped on one of the dongles on band "Ch1 423 MHz".
But the sound of the radio station at 423.650 MHz is still heard normally:
http://kiev.extmail.info/#freq=423650000,mod=nfm,sql=-150

Switching bands, refreshing the page, and changing the browser doesn't help.

On the other five bands, the waterfalls continue to function normally.

I don't change anything and leave it unchanged for you to see for yourself.

Logs for today are attached.

Best regards, Shilov

[daemon.tar.gz](https://github.com/jketterl/openwebrx/files/4976327/daemon.tar.gz)

jketterl, 2020-07-25:
The attachement does not contain anything related to OpenWebRX. Either the logs have been directed elsewhere, or there has been no events (unlikely).

P.S. you don't need to leave things untouched. Remote diagnosis is usually not possible by looking at the web frontend.

T-Shilov, 2020-07-25:
Ок, what should I do next?

jketterl, 2020-07-25:
a restart will probably clear the problem for now. without any information to work on, however, I cannot diagnose any further or take measures to prevent this from happening again.

T-Shilov, 2020-07-25:
> a reboot will probably fix the problem

Yes, I agree with you. But it will not find the cause of the problem.

Specify, please, what information you need. I will try to provide it.

jketterl, 2020-07-25:
Well, I don't know your setup. I would need the output of the openwebrx process, but it's difficult for me to know where that would be without knowing the way the setup was done.

- If you installed from packages (or used a Raspberry Pi image), OpenWebRX is set up to be run by systemd. The output is collected and made available by `sudo journalctl -u openwebrx`.
- If you're running the docker container images, it's down to the docker setup where the output goes. If there's no specific provisions in place, you will likely be able to get the output with the `docker logs` command.
- If you followed the manual installation procedures, it's completely up to you where those logs end up. There is an optional paragraph on how to set up OpenWebRX with systemd, if you followed that, you should be able to use `sudo journalctl -u openwebrx` as well.
- In case of any modifications to any of the above, you should probably know yourself where the output goes.

T-Shilov, 2020-07-25:
Please, here's my details:

1. I installed OpenWebRX v.0.19.1 according to this instruction:

https://www.openwebrx.de/download/debian.php

```
wget -O - https://repo.openwebrx.de/debian/key.gpg.txt | apt-key add
echo "deb https://repo.openwebrx.de/debian/ buster main" > /etc/apt/sources.list.d/openwebrx.list
apt-get update
apt-get install openwebrx
```

2. I did not make any changes to the code of your application, except for the photo and the name of the my site.

That's all, actually.

jketterl, 2020-07-25:
Well it's not really helpful to know that now; and please don't expect me to remember it. There's [four different installation routines](https://github.com/jketterl/openwebrx/wiki/Setup-Guide) available on the Wiki, and there's hundreds of installations, and I cannot guess how many of them have been modified. I cannot anticipate it.

The focus should be on collecting the logs. Reiterating from my previous answer, you should be able to obtain them using `sudo journalctl -u openwebrx`.

T-Shilov, 2020-07-25:
Command gives blank:

`# journalctl -u openwebrx `

This command provided the data that is in the archive:

`# journalctl > journalctl.txt`

[journalctl.tar.gz](https://github.com/jketterl/openwebrx/files/4976785/journalctl.tar.gz)

jketterl, 2020-07-25:
It is blank indeed. But the logfile also shows why: You're running OpenWebRX through cron, with it's output redirected to `/dev/null`.

`июл 25 19:21:58 webrx CRON[481]: (root) CMD (sleep 10 && /usr/bin/openwebrx >/dev/null 2>&1)`

If that's how your receiver has been running when the waterfall cut out, it is pretty clear that no logs can be recovered. The redirect makes sure that any output is lost.

Further, I would clearly consider this a modification. :angry: 

Even further, I do not see the point in starting OpenWebRX from cron. SystemD has very good capabilities to ensure that a unit stays up and running.

closed 2020-07-25 by jketterl

T-Shilov, 2020-07-25:
Jacob, I will now explain why I have to launch OpenWebrH via Cron.

jketterl, 2020-07-25:
Don't. It doesn't matter. You should have known and have responded accordingly when I was explicitly inquiring for modifications. You should have realized right from the beginning that there was nothing to recover with that redirection in place.

Your station setup is your business. But please don't make me find out all the details this way.

T-Shilov, 2020-07-25:
So, this my solution is caused by another problem:

**OpenWebRX suddenly stopped working**
https://github.com/jketterl/openwebrx/issues/146

You explained it with the following reason: "too many open files".

Aadditionally  is a second problem:
**Not all dongles are started # 150**
https://github.com/jketterl/openwebrx/issues/150

Since these two problems have not yet been resolved, I have to perform a daily reboot of the system via Cron and a delayed start of the OpenWebRX service for 10 seconds.

This was a forced and temporary solution.
When you fix these two problems, I will disable the daily reboot and disable start via Cron.

jketterl, 2020-07-25:
Look, I don't know why you're trying to blame me now. You're the one that modified the receiver, so you're the one that needs to know where the output goes. It's not really that hard, but instead of at least being honest when I laid out the options, you let me dig in your logfiles to find out that you have presented me with incorrect information.

I don't need to know your modifications. I don't even want to know your modifications. All that matters is that once you take things into your hand, you're the one responsible for locating and providing the information that is needed to fix problems. You have not been very helpful at that, and I lost a lot of time that way.

T-Shilov, 2020-07-25:
Because I'm not a developer. And I'm not even an experienced user. I am a regular Linux user and have a modest knowledge of Linux.
But I do my best to help your project by observing its work.
But do not demand from me too much knowledge of what and how to do. I do not have this knowledge.

Best regards, Shilov.

jketterl, 2020-07-25:
No, it's because you're asking too much from yourself. You're trying to fix problems that you cannot fix. You're trying to make this the perfect receiver, and that stupid developer doesn't fix the problems that you encounter, so you're trying to patch them yourself, without knowing the consequences.

I can totally understand, but please don't lose your head, and if you're not sure, get somebody to help you out.