jketterl/openwebrx #158: a bug when only config with one SDR device

d9394, 2020-08-03:
2020-08-03 09:19:50,563 - owrx.__main__ - INFO - OpenWebRX version v0.20.0-dev starting up...
2020-08-03 09:19:50,661 - owrx.sdr - INFO - SDR sources loaded. Available SDRs: SDRPlay RSP2
2020-08-03 09:19:57,147 - owrx.connection - DEBUG - client connection intitialized
2020-08-03 09:19:57,175 - owrx.source - DEBUG - activating profile 20m
2020-08-03 09:19:58,476 - owrx.source - INFO - Started sdr source: soapy_connector -d driver=sdrplay -p 60145 -a "Antenna A" -c 36455 -s 500000 -f 14150000 -P 0 -g 0
Tuner gain set to 0.00 dB.
IQ worker thread started
socket setup complete, waiting for connections
[INFO] Using format CS16.
setting up control socket...
control socket started on 36455
client connection establised
2020-08-03 09:20:00,199 - owrx.source.connector - DEBUG - opening control socket...
control connection established
2020-08-03 09:20:00,201 - owrx.dsp - DEBUG - received STATE_RUNNING, attempting DspSource restart
2020-08-03 09:20:00,235 - owrx.bands - WARNING - Frequency for wspr on 4m is not within band limits: 70091000
2020-08-03 09:20:00,268 - owrx.fft - DEBUG - Spectrum thread initialized successfully.
2020-08-03 09:20:00,269 - csdr.csdr - DEBUG - Command = nc -v 127.0.0.1 60145 | csdr fft_cc 4096 2923.9766081871344 | csdr logaveragepower_cf -70 4096 19 | csdr fft_exchange_sides_ff 4096 | csdr compress_fft_adpcm_f_u8 4096
client connection establised
Connection to 127.0.0.1 60145 port [tcp/*] succeeded!
closing client socket
2020-08-03 09:20:01,923 - csdr.csdr - DEBUG - Command = nc -v 127.0.0.1 60145 | csdr shift_addfast_cc --fifo /tmp/openwebrx_pipe_1972954608_shift_pipe | csdr fir_decimate_cc 45 0.0033333333333333335 HAMMING | csdr bandpass_fir_fft_cc --fifo /tmp/openwebrx_pipe_1972954608_bpf_pipe 0.0288 HAMMING | csdr squelch_and_smeter_cc --fifo /tmp/openwebrx_pipe_1972954608_squelch_pipe --outfifo /tmp/openwebrx_pipe_1972954608_smeter_pipe 5 1 | csdr realpart_cf | csdr fractional_decimator_ff 1.0078105316200554 | csdr agc_ff | csdr limit_ff | csdr convert_f_s16 | csdr encode_ima_adpcm_i16_u8
2020-08-03 09:20:01,939 - owrx.dsp - DEBUG - adding new output of type audio
csdr shift_addfast_cc: fifo control mode on
2020-08-03 09:20:01,954 - owrx.dsp - DEBUG - adding new output of type smeter
client connection establised
csdr bandpass_fir_fft_cc: csdr shift_addfast_cc: reinitialized to 0.16
Connection to 127.0.0.1 60145 port [tcp/*] succeeded!
fifo control mode on
fir_decimate_cc: taps_length = 1199
csdr squelch_and_smeter_cc: fifo control mode on
csdr fir_decimate_cc: taps_length = 1199
csdr fir_decimate_cc: padded_taps_length = 1200
csdr fir_decimate_cc: taps = 7a018
csdr fir_decimate_cc: NEON aligned taps = 7a020
csdr bandpass_fir_fft_cc: (fft_size = 512) = (taps_length = 139) + (input_size = 374) - 1
(overlap_length = 138) = taps_length - 1
csdr squelch_and_smeter_cc: initial squelch level is 1e-15
csdr fractional_decimator_ff: use_prefilter = 0, num_poly_points = 12, transition_bw = 0.03, window = HAMMING
csdr fractional_decimator_ff: not using taps
csdr bandpass_fir_fft_cc: filter initialized, low_cut = 0.027, high_cut = 0.27
2020-08-03 09:20:06,984 - owrx.websocket - ERROR - Exception in websocket handler handleTextMessage()
Traceback (most recent call last):
  File "/opt/openwebrx/owrx/websocket.py", line 206, in read_loop
    self.messageHandler.handleTextMessage(self, message)
  File "/opt/openwebrx/owrx/connection.py", line 175, in handleTextMessage
    self.sdr.activateProfile(profile[1])
IndexError: list index out of range

my config_webrx.py:

sdrs = {
    """
    "rtlsdr": {
        "name": "RTL-SDR USB Stick",
        "type": "rtl_sdr",
        "ppm": 0,
        # you can change this if you use an upconverter. formula is:
        # center_freq + lfo_offset = actual frequency on the sdr
        "lfo_offset": 50000000,
        "profiles": {
            "70cm": {
                "name": "70cm Relais",
                "center_freq": 438800000,
                "rf_gain": 30,
                "samp_rate": 2400000,
                "start_freq": 439275000,
                "start_mod": "nfm",
            },
            "2m": {
                "name": "2m komplett",
                "center_freq": 145000000,
                "rf_gain": 30,
                "samp_rate": 2400000,
                "start_freq": 145725000,
                "start_mod": "nfm",
            },
        },
    },
    "airspy": {
        "name": "Airspy HF+",
        "type": "airspyhf",
        "ppm": 0,
        "rf_gain": "auto",
        "profiles": {
            "20m": {
                "name": "20m",
                "center_freq": 14150000,
                "samp_rate": 384000,
                "start_freq": 14070000,
                "start_mod": "usb",
            },
            "30m": {
                "name": "30m",
                "center_freq": 10125000,
                "samp_rate": 192000,
                "start_freq": 10142000,
                "start_mod": "usb",
            },
            "40m": {
                "name": "40m",
                "center_freq": 7100000,
                "samp_rate": 256000,
                "start_freq": 7070000,
                "start_mod": "lsb",
            },
            "80m": {
                "name": "80m",
                "center_freq": 3650000,
                "samp_rate": 384000,
                "start_freq": 3570000,
                "start_mod": "lsb",
            },
            "49m": {
                "name": "49m Broadcast",
                "center_freq": 6050000,
                "samp_rate": 384000,
                "start_freq": 6070000,
                "start_mod": "am",
            },
        },
    },
    """
    "sdrplay": {
        "name": "SDRPlay RSP2",
        "type": "sdrplay",
        "ppm": 0,
        "antenna": "Antenna A",
        "profiles": {
            "20m": {
                "name": "20m",
                "center_freq": 14150000,
                "rf_gain": 0,
                "samp_rate": 500000,
                "start_freq": 14070000,
                "start_mod": "usb",
            },
            "30m": {
                "name": "30m",
                "center_freq": 10125000,
                "rf_gain": 0,
                "samp_rate": 250000,
                "start_freq": 10142000,
                "start_mod": "usb",
            },
            "40m": {
                "name": "40m",
                "center_freq": 7100000,
                "rf_gain": 0,
                "samp_rate": 500000,
                "start_freq": 7070000,
                "start_mod": "lsb",
            },
            "80m": {
                "name": "80m",
                "center_freq": 3650000,
                "rf_gain": 0,
                "samp_rate": 500000,
                "start_freq": 3570000,
                "start_mod": "lsb",
            },
            "49m": {
                "name": "49m Broadcast",
                "center_freq": 6000000,
                "rf_gain": 0,
                "samp_rate": 500000,
                "start_freq": 6070000,
                "start_mod": "am",
            },
        },
    },
}

jketterl, 2020-08-03:
Using code blocks really helps keeping things readable :wink: 

I cannot reproduce the error, at least not with latest develop, the config works fine. Do you have any modifications?

jketterl, 2020-08-05:
You know what, scratch that. I did copy your example, but I already left out the triple quoted part, thinking it's just comments. At that point I was already making pretty much the same mistake, just the other way round.

Python does not have a way of commenting multiple lines, as many other languages do. The common practice of using triple quotes works in most cases, simply because a string literal without any assignment has no further meaning in code. The sdr configuration, however, is a dictionary, and a the triple quoted string at this position gets concatenated to the first key ("sdrplay"). This breaks the way the profile selection dropdown works since it is not prepared to accept multiple lines of json as its value. This happens behind the scenes, and is only visible using the browser's developer tools.

Easy fix: simply remove the superfluous lines instead of triple quoting them. If you want to keep them for later experiments, the simplest way of disabling one of the sdr devices is by selecting an invalid type (e.g. by prefixing it with `disabled_`). Doesn't sound very convincing, but the feature check will make sure very early that such configs are filtered out. I've used this myself on many occasions.

d9394, 2020-08-05:
Roger, use 'disabled_rtl' as sdr type config can be ingore by system startup.

closed 2020-08-05 by d9394