jketterl/openwebrx #159: Possibility listen themselves

T-Shilov, 2020-08-04:
Dear Jakob,

I ask you to make in the configuration file a parameter for an optional delay of the audio signal from 0 to 20 seconds.
This is very important for radio amateurs so that they can listen to themselves and find out what their transmitter sound quality is.
Something like it happens on the "parrot" radio repeaters.
This is just a wish.

Best ragards,
        Shilov

jketterl, 2020-08-04:
I don't think you thought this through. You certainly don't want to set a config that affects all users, just in case a HAM wants do some testing on your receiver. The latency and lack of responsiveness is already an issue with OpenWebRX.

You obviously also didn't considert that delaying the audio by 20s will also mean that your inputs (changing the modulator, changing the freqency, ...) will take 20s to have an audible effect.

This is one of those cases where I must say: This is hurting the primary use case of OpenWebRX to an inacceptible extend, and won't become reality.

T-Shilov, 2020-08-04:
The situation is the opposite. This was taken into account long ago, and for some time I used this opportunity in another similar project (WebSDR).
All HAM were pleased because they could check the quality of their equipment at any time and it really helped them.
The value of the broadcast delay can be set depending on them wishes.

But I didn’t come up with it. I didn’t come up with anything new, and just repeated the idea that had already been implemented earlier.
This is how the parameter is described in the configuration file
https://github.com/FarnhamSDR/websdr/blob/master/dist11/doc/config.txt :

> **delay x**
>   Artificially delay signals by x seconds.
>   Useful to prevent cheating in contests, if your receiver is so special   that using it would give participants an unfair advantage (e.g., a very big dish for EME).

But you can implement this idea even better than in the WebSDR.
To do this, you need to set the desired delay not in the configuration file, but let HAM themselves set the desired delay or turn it off with a completely corresponding buttons on the OpenWebRX control panel.

This is, of course, more difficult, but much better.

jketterl, 2020-08-04:
Great, another websdr feature. My spirits drop to the floor every time one of them swaps over here... I haven't seen this one in action, probably you'd only turn it on while you use it.

Anyway, when implemented as a client-side control, it won't affect others, but will be somewhat more complicated, mostly because there is no space in the control panel for such specific use cases (ease of use clearly is more significant here). That means it will need it's own controls, in a place that is not even available at this point.

This is pretty much the same level as the "private mode" use case, that means it's low priority and will need to wait.

T-Shilov, 2020-08-04:
Jakob, I only politely reminded you of the existence of this useful feature.
To do it or not to do it is up to you, of course.