jketterl/openwebrx #160: Add a keyboard interface

f6fvy, 2020-08-07:
Hi

I'd like to use a MIDI controller with encoders to control the VFO freq, and various push buttons to control options. As it probably won't be possible to interface a MIDI device directly, the easiest way would be to send keystrokes on MIDI events (this kind of utility already exists).

Unfortunately, unless I missed something, it seems that OpenWebRX doesn't allow any keystroke commands. Would it be possible to consider adding them ? If so, please, be inspired by the current assignments of the websdr server to unify interface...

Thanks !

jketterl, 2020-08-07:
I'll just have to pile this up with all the other feature requests, sorry, but work pace is currently slow due to many open issues and lack of inspiration.

I cannot promise to take the inspiration from websdr, it's usually not a very great example, but then again I don't know what the keyboard shortcuts look like.

f6fvy, 2020-08-07:
For the record

[image]

Even if all keystrokes don't even make sense in the OpenWebRX context, the j/k/arrows + optional modifiers (Shift / Control/ Alt) keys to control the RX freq would be a very good start. Thanks for your consideration. 73 !

jketterl, 2020-08-07:
I'll have a look. Doesn't seem too much, some of them don't apply: VFO A/B, waterfall works differently, and the mode buttons don't cover all of them, maybe a modifier for those...

I don't really intend to carry the websdr legacy, I'd much prefer to move on for the sake of improvement, so I do keep my reservations in such cases, I hope you understand.

BTW, the idea of keyboard control has been on the list for some time now. I'm currently looking into setting up issue templates for the project so that the project gets more focus.

bhall7, 2024-08-18:
I would also like to add this to my wish list. As I use OWRX, I’d love to be able to use the keys, "a A d l u c f q" to toggle modes (AM SAM DRM LSB USB CW NBFM IQ).