jketterl/openwebrx #162: curl_json

loewal, 2020-08-11:
Hi Jakob,

I followed your instructions "How to get openwebrx stats into collectd",
but I keep getting errors:

plugin_load: Could not find plugin "curl_json" in /usr/lib/collectd
aug 11 11:46:30 Pi4 collectd[1415]: There is configuration for the `curl_json' plugin, but the plugin isn't loaded. Please check your configuration.
Does it need to be installed somehow?

loewal, 2020-08-11:
And another thing...this command didn't work: sudo apt-get build-dep collectd
root@Pi4:~ # sudo apt-get build-dep collectd
Pakketlijsten worden ingelezen... Klaar
E: Uw bronnenlijst (/etc/apt/sources.list) dient tenminste één bron-URI te bevatten
I hope you can understand this Dutch....

jketterl, 2020-08-11:
Sorry, but I have set up the issue templates to prevent these things from popping up on here, this is not an issue with OpenWebRX. I will handle these things more strictly because the issue tracker is not the place for support and questions.

If you need help setting up collectd, feel free to visit our mailing list.

closed 2020-08-11 by jketterl