jketterl/openwebrx #163: SDRPlay RSP1 not recognized

StaRipper, 2020-08-11:
**Describe the bug**
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
2020-08-11 17:21:17,932 - owrx.__main__ - INFO - OpenWebRX version v0.19.1 starting up...
2020-08-11 17:21:18,411 - owrx.sdr - INFO - SDR sources loaded. Available SDRs: RTL-SDR USB Stick, Airspy HF+, SDRPlay RSP2
2020-08-11 17:21:41,135 - owrx.connection - DEBUG - client connection intitialized
2020-08-11 17:21:41,158 - owrx.source - DEBUG - activating profile 70cm
2020-08-11 17:21:41,172 - owrx.source - DEBUG - activating profile 20m
2020-08-11 17:21:41,179 - owrx.source - DEBUG - activating profile 20m
2020-08-11 17:21:42,360 - owrx.source - INFO - Started sdr source: rtl_connector -g 30 -s 2400000 -P 0 -f 438800000 -p 46721 -c 38365
No supported devices found.
no device found.
2020-08-11 17:21:42,381 - owrx.source - DEBUG - shut down with RC=1
2020-08-11 17:21:42,463 - owrx.source.connector - DEBUG - opening control socket...
2020-08-11 17:21:42,466 - owrx.source - ERROR - Exception during postStart()
Traceback (most recent call last):
  File "/opt/openwebrx/owrx/source/__init__.py", line 195, in start
    self.postStart()
  File "/opt/openwebrx/owrx/source/connector.py", line 52, in postStart
    self.controlSocket.connect(("localhost", self.controlPort))
ConnectionRefusedError: [Errno 111] Connection refused
2020-08-11 17:21:42,468 - owrx.dsp - DEBUG - received STATE_FAILED, shutting down DspSource
2020-08-11 17:21:42,469 - owrx.connection - WARNING - SDR device "RTL-SDR USB Stick" has failed, selecing new device
2020-08-11 17:21:43,656 - owrx.source - INFO - Started sdr source: soapy_connector -g auto -s 384000 -P 0 -f 14150000 -p 36901 -c 58211 -d driver=airspyhf
SoapySDRDevice_make failed
2020-08-11 17:21:43,705 - owrx.source - DEBUG - shut down with RC=1
2020-08-11 17:21:43,759 - owrx.source.connector - DEBUG - opening control socket...
2020-08-11 17:21:43,760 - owrx.source - ERROR - Exception during postStart()
Traceback (most recent call last):
  File "/opt/openwebrx/owrx/source/__init__.py", line 195, in start
    self.postStart()
  File "/opt/openwebrx/owrx/source/connector.py", line 52, in postStart
    self.controlSocket.connect(("localhost", self.controlPort))
ConnectionRefusedError: [Errno 111] Connection refused
2020-08-11 17:21:43,761 - owrx.dsp - DEBUG - received STATE_FAILED, shutting down DspSource
2020-08-11 17:21:43,762 - owrx.connection - WARNING - SDR device "Airspy HF+" has failed, selecing new device
2020-08-11 17:21:44,921 - owrx.source - INFO - Started sdr source: soapy_connector -g 0 -s 500000 -a "Antenna A" -P 0 -f 14150000 -p 47711 -c 51869 -d driver=sdrplay
libusb: error [_get_usbfs_fd] libusb couldn't open USB device /dev/bus/usb/001/005: No such device
[WARNING] Can't find label in args
SoapySDRDevice_make failed
2020-08-11 17:21:45,987 - owrx.source - DEBUG - shut down with RC=1
2020-08-11 17:21:46,032 - owrx.source.connector - DEBUG - opening control socket...
2020-08-11 17:21:46,034 - owrx.source - ERROR - Exception during postStart()
Traceback (most recent call last):
  File "/opt/openwebrx/owrx/source/__init__.py", line 195, in start
    self.postStart()
  File "/opt/openwebrx/owrx/source/connector.py", line 52, in postStart
    self.controlSocket.connect(("localhost", self.controlPort))
ConnectionRefusedError: [Errno 111] Connection refused
2020-08-11 17:21:46,035 - owrx.dsp - DEBUG - received STATE_FAILED, shutting down DspSource
2020-08-11 17:21:46,036 - **owrx.connection - WARNING - SDR device "SDRPlay RSP2" has failed, selecing new device**
2020-08-11 17:21:46,038 - owrx.connection - WARNING - no more SDR devices available
2020-08-11 17:21:47,365 - owrx.bands - WARNING - Frequency for wspr on 4m is not within band limits: 70091000
2020-08-11 17:21:47,391 - owrx.property - ERROR - unsupported operand type(s) for +: 'int' and 'NoneType'
Traceback (most recent call last):
  File "/opt/openwebrx/owrx/property/__init__.py", line 78, in _fireCallbacks
    c.call(name, value)
  File "/opt/openwebrx/owrx/property/__init__.py", line 17, in call
    self.subscriber(*args, **kwargs)
  File "/opt/openwebrx/owrx/dsp.py", line 73, in set_dial_freq
    freq = self.props["center_freq"] + self.props["offset_freq"]
TypeError: unsupported operand type(s) for +: 'int' and 'NoneType'

**To Reproduce**
Start the app with SDRPlay RSP1.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Installation method**
Docker image on Raspberry PI 3

**Versions**
latest

jketterl, 2020-08-11:
Well, I don't have an RSP1, so I cannot test with that. But I run an RSP2 with a RPi docker setup, similar to yours, and it fails to start up successfully quite often, too. I'm not sure what exactly is happening, but I guess it's somewhere between docker, the USB subsystem and maybe the hardware. The most plausible cause is insufficient power supply from the Pi since similar things happen no non-docker setups.

From your messages:

> libusb: error [_get_usbfs_fd] libusb couldn't open USB device /dev/bus/usb/001/005: No such device

That tells me that the device initially was there, was forwarded to the container, and then moved or disappeared. As far as I know, the device file system is not synced between the host and the container, so basically at this point, the container is lacking the correct device file to be able to open the device.

nmaster2042, 2020-08-14:
I don't know if it's related, but using docker image on a arm64 SBC, when I plug a RSP device and then starts openwebrx image I get this issue. I restart the container and it works.

I have this issue since using SDRPlay v 3 API.

It seems the first time docker container is started, and API service is started, something is happening and makes RSP to change usb bus id.

When starting a second time, no more usb bus id change and RSP can be initialised / used.

I can start / stop and restart the docker container without issue until un unplug and re plug the device or reboot SBC.

Note that  have the same issue withe other docker images made by myself, only on SBCs, not X86_64 computer.

jketterl, 2020-08-14:
Yes, as I said, there is something odd happening with the USB devices that the device forwarding of docker cannot cope with.

Since this is not an issue in OpenWebRX, I will close this.

closed 2020-08-14 by jketterl