jketterl/openwebrx #164: Docker - Direwolf fails due to missing libhamlib.so.2

zand, 2020-08-14:
**Describe the bug**
Direwolf fails to start due to missing libhamlib.so.2 in docker image.

**To Reproduce**
Select Packet as the demod type.

**Expected behavior**
For OpenWebRX to decode and display APRS packets. 

**Installation method**
Docker image: jketterl/openwebrx:stable

**Versions**
What version of OpenWebRX are you running? (Check on startup, or see `owrx/version.py`. If a `-dev` version is used, ideally state the commit the issue is appearing on)

**Log messages**
```
2020-08-14 04:47:34,391 - csdr.csdr - DEBUG - dsp thread ended with rc=-15
2020-08-14 04:47:34,393 - csdr.csdr - DEBUG - Command = nc -v 127.0.0.1 55033 | csdr shift_addfast_cc --fifo /tmp/openwebrx/openwebrx_pipe_140549994934456_shift_pipe | csdr fir_decimate_cc 50 0.003 HAMMING | csdr bandpass_fir_fft_cc --fifo /tmp/openwebrx/openwebrx_pipe_140549994934456_bpf_pipe 0.006666666666666667 HAMMING | csdr squelch_and_smeter_cc --fifo /tmp/openwebrx/openwebrx_pipe_140549994934456_squelch_pipe --outfifo /tmp/openwebrx/openwebrx_pipe_140549994934456_smeter_pipe 5 8 | csdr tee /tmp/openwebrx/openwebrx_pipe_140549994934456_iqtee_pipe | csdr tee /tmp/openwebrx/openwebrx_pipe_140549994934456_iqtee2_pipe | csdr fmdemod_quadri_cf | csdr limit_ff | csdr deemphasis_nfm_ff 48000 | sox -t raw -r 48000 -e floating-point -b 32 -c 1 --buffer 32 - -t raw -r 11025 -e signed-integer -b 16 -c 1 -  | csdr encode_ima_adpcm_i16_u8
2020-08-14 04:47:34,398 - owrx.dsp - DEBUG - adding new output of type audio
2020-08-14 04:47:34,399 - csdr.csdr - DEBUG - starting secondary demodulator from IF input sampled at 48000
csdr shift_addfast_cc: fifo control mode on
csdr bandpass_fir_fft_cc: fifo control mode on
csdr squelch_and_smeter_cc: fifo control mode on
Connection to 127.0.0.1 55033 port [tcp/*] succeeded!
client connection establised
fir_decimate_cc: taps_length = 1333
2020-08-14 04:47:34,401 - csdr.csdr - DEBUG - secondary command (demod) = cat /tmp/openwebrx/openwebrx_pipe_140549994934456_iqtee2_pipe | csdr fmdemod_quadri_cf | csdr convert_f_s16 | direwolf -c /tmp/openwebrx/openwebrx_direwolf_140549994934456.conf -r 48000 -t 0 -q d -q h 1>&2
2020-08-14 04:47:34,401 - csdr.csdr - DEBUG - secondary command (fft) = cat /tmp/openwebrx/openwebrx_pipe_140549994934456_iqtee_pipe | csdr realpart_cf | csdr fft_fc 1024 4800.0 | csdr logpower_cf -70 | csdr compress_fft_adpcm_f_u8 1024
2020-08-14 04:47:34,405 - owrx.dsp - DEBUG - adding new output of type secondary_fft
csdr tee: file opened: /tmp/openwebrx/openwebrx_pipe_140549994934456_iqtee_pipe
csdr shift_addfast_cc: csdr bandpass_fir_fft_cc: (fft_size = 1024) = (taps_length = 599) + (input_size = 426) - 1
(overlap_length = 598) = taps_length - 1
reinitialized to 0.254167
csdr squelch_and_smeter_cc: initial squelch level is 1e-15
csdr tee: file opened: /tmp/openwebrx/openwebrx_pipe_140549994934456_iqtee2_pipe
direwolf: error while loading shared libraries: libhamlib.so.2: cannot open shared object file: No such file or directory
closing client socket
csdr bandpass_fir_fft_cc: filter initialized, low_cut = -0.0833333, high_cut = 0.0833333
```

jketterl, 2020-08-14:
I can confirm, direwolf is linked against hamlib, which is stripped from the docker images in an effort to make them as slim as possible. Thanks for reporting this.

I will see if I can get this set up to work without hamlib, there's no big point in having it for the operation of OpenWebRX.

jketterl, 2020-08-14:
Alright, that wasn't too bad. A simple argument during compilation prevents direwolf from using hamlib altogether.

The change is in the `:latest` images now, if you don't mind you could switch over for a quick tesṭ.

zand, 2020-08-15:
Just got done testing :latest and can confirm that it is working.

I would like to thank you for the fast response and taking your time to work on OpenWebRX.