jketterl/openwebrx #166: Unable to set-up APRS Igate

f5bcb, 2020-08-18:
Hi,
Starting from the last image (2020-06-13-OpenWebRX-full.zip) on Raspi4 , then following configuration instruction, I'm facing a problem about the APRS Igate despite the multiple checking of the main config file.
The APRS is working but the Igate seem dead : no report to APRS IS.

The Direwolf process is running 👍 

pi@openwebrx:~ $ ps -edf | grep direwolf
openweb+  1425   807  0 16:17 ?        00:00:00 /bin/sh -c cat /tmp/openwebrx/openwebrx_pipe_3039929520_iqtee2_pipe | csdr fmdemod_quadri_cf | csdr fractional_decimator_ff 1.0101010101010102 | csdr convert_f_s16 | direwolf -c /tmp/openwebrx/openwebrx_direwolf_3039929520.conf -r 48000 -t 0 -q d -q h 1>&2
openweb+  1430  1425  2 16:17 ?        00:00:00 direwolf -c /tmp/openwebrx/openwebrx_direwolf_3039929520.conf -r 48000 -t 0 -q d -q h
pi        1441  1120  0 16:17 pts/1    00:00:00 grep --color=auto direwolf
pi@openwebrx:~ $

Is there some log file to check ? No Direwolf log found :-(

Thanks a lot for any suggestion
Jeff, F5BCB

jketterl, 2020-08-18:
Well, there was instructions on the Issue template that you obviously didn't bother with: On most installations, the log should be available using the command `sudo journalctl -u openwebrx` - that includes what direwolf puts out.

The APRS reporting is only available for background services. That change was necessary because the APRS-IS network does only allow one connection per IP, and the reporting is directly handled by direwolf. Are you running the APRS decoding as a background service?

Did you check if the APRS frequency matches what's in use in your area? The default that comes with OpenWebRX is 144.8, but from what I know that doesn't apply everywhere.

f5bcb, 2020-08-18:
Thanks for your comments.

A/ - Issuing the journalctl command you suggest do not allow to get somme usefull informations on Direwolf and what he send or receive 👍 
pi@openwebrx:/etc/openwebrx $ sudo journalctl -u openwebrx | grep direwolf
Aug 18 18:15:17 openwebrx openwebrx[806]: 2020-08-18 18:15:17,383 - csdr.csdr - DEBUG - secondary command (demod) = cat /tmp/openwebrx/openwebrx_pipe_3043217072_iqtee2_pipe | csdr fmdemod_quadri_cf | csdr fractional_decimator_ff 1.0101010101010102 | csdr convert_f_s16 | direwolf -c /tmp/openwebrx/openwebrx_direwolf_3043217072.conf -r 48000 -t 0 -q d -q h 1>&2
Aug 18 18:15:17 openwebrx openwebrx[806]: Reading config file /tmp/openwebrx/openwebrx_direwolf_3043217072.conf
Aug 18 18:15:53 openwebrx openwebrx[806]: 2020-08-18 18:15:53,024 - csdr.csdr - DEBUG - secondary command (demod) = cat /tmp/openwebrx/openwebrx_pipe_3043217072_iqtee2_pipe | csdr fmdemod_quadri_cf | csdr fractional_decimator_ff 1.0101010101010102 | csdr convert_f_s16 | direwolf -c /tmp/openwebrx/openwebrx_direwolf_3043217072.conf -r 48000 -t 0 -q d -q h 1>&2
Aug 18 18:15:53 openwebrx openwebrx[806]: Reading config file /tmp/openwebrx/openwebrx_direwolf_3043217072.conf
pi@openwebrx:/etc/openwebrx $

B/- If the background service is set by "services_enabled = True", the response is NO, this parameter is set to False at this time
I already set to "True" without more success.
Moreover, I read a sample "config_webrx.py" file at this URL : https://tinyurl.com/yylfqbgm and it seem that the igate do not require to set "service_enabled".

C/ - Local APRS frequency is 144,800 mHz.

jketterl, 2020-08-18:
You'll need the services enabled. I don't know why you'd think quoting a random config (that shouldn't be online, it contains passwords...) would convince me of the opposite... If you don't trust me, check the code, or open the config file passed to direwolf while it's running: you'll find no `IGSERVER` or `IGLOGIN` lines there. I've even given you the reason why this restriction has been implemented.